slic3r/Slic3r #3582: No description of interface layers in manual

mrjohnc, 2016-11-22:
There is no description of what interface layers are or do in the Slic3r manual

http://manual.slic3r.org/expert-mode/support-material

lordofhyphens, 2016-11-27:
... they're the number of layers in the interface (sparse print between the part and proper support). I thought that was obvious from the name, but that could just be my own thought processes :)

lordofhyphens, 2017-01-26:
https://github.com/alexrj/Slic3r-Manual

I'd appreciate it a lot if you opened a separate issue against the manual's repository (even better, write a description and issue a pull request).

timur-tabi, 2019-01-30:
Was a Slic3r-Manual ticket ever opened?  I don't see any description of Interface Layers in the manual.  In fact, the entire "Support Material" section is very sparse.

mrjohnc, 2019-01-30:
I didn't do it because I don't know how to do it or what an interface layer is