slic3r/Slic3r #3583: latest changes: cant compile under windows

Tinchus2009, 2016-11-23:
### Version
latest version

### Operating system type + version
win10 64

### Behavior
An hour ago, I issued the normal GIT PULL, PERL BUILD.PL commands
In the build i get to this point:
t\22_config.t ................... ok

And it stays there for ever. I gave it like 15 minutes and just stayed there.
The git pull I did 1 week ago apart from the opengl problem, builded with no problems.

lordofhyphens, 2016-11-23:
Build server already is reporting failure on Windows, it will be investigated.

bubnikv, 2016-11-23:
That is the purpose of the test case, to hang on a broken strawberry perl
64bit.

lordofhyphens, 2016-11-23:
@bubnikv right, derp. I didn't make the connection  (as the build server
which is 32bit is passing). I wonder if we could write in a timeout to fail
more gracefully.

bubnikv, 2016-11-23:
I don't know whether you could write a timeout, but I have changed couple
of places in Slic3r to not use the deadly C++ try / catch -> perl exception
combination, and I have added a compile switch, which disables this test
case. Currently our 64bit builds are compiled with a buggy Strawberry and
it seems to work fine.

Tinchus2009, 2016-11-27:
I just have issued a GIT PULL command, now there are errors reported in the build:

C:\dev\boost_1_59_0/boost/bind/bind.hpp:467:40: error: macro "PerlSock_accept" requires 3 arguments, but only 1 given
     template<class V> void accept(V & v) const
                                        ^
C:\dev\boost_1_59_0/boost/bind/bind.hpp:469:28: error: macro "PerlSock_accept" requires 3 arguments, but only 1 given
         base_type::accept(v);
                            ^
C:\dev\boost_1_59_0/boost/bind/bind.hpp:535:40: error: macro "PerlSock_accept" requires 3 arguments, but only 1 given
     template<class V> void accept(V & v) const
                                        ^
C:\dev\boost_1_59_0/boost/bind/bind.hpp:537:28: error: macro "PerlSock_accept" requires 3 arguments, but only 1 given
         base_type::accept(v);
                            ^
C:\dev\boost_1_59_0/boost/bind/bind.hpp:606:40: error: macro "PerlSock_accept" requires 3 arguments, but only 1 given
     template<class V> void accept(V & v) const
                                        ^
C:\dev\boost_1_59_0/boost/bind/bind.hpp:608:28: error: macro "PerlSock_accept" requires 3 arguments, but only 1 given
         base_type::accept(v);
                            ^
C:\dev\boost_1_59_0/boost/bind/bind.hpp:680:40: error: macro "PerlSock_accept" requires 3 arguments, but only 1 given
     template<class V> void accept(V & v) const
                                        ^
C:\dev\boost_1_59_0/boost/bind/bind.hpp:682:28: error: macro "PerlSock_accept" requires 3 arguments, but only 1 given
         base_type::accept(v);
                            ^
C:\dev\boost_1_59_0/boost/bind/bind.hpp:757:40: error: macro "PerlSock_accept" requires 3 arguments, but only 1 given
     template<class V> void accept(V & v) const
                                        ^
C:\dev\boost_1_59_0/boost/bind/bind.hpp:759:28: error: macro "PerlSock_accept" requires 3 arguments, but only 1 given
         base_type::accept(v);
                            ^
C:\dev\boost_1_59_0/boost/bind/bind.hpp:837:40: error: macro "PerlSock_accept" requires 3 arguments, but only 1 given
     template<class V> void accept(V & v) const
                                        ^
C:\dev\boost_1_59_0/boost/bind/bind.hpp:839:28: error: macro "PerlSock_accept" requires 3 arguments, but only 1 given
         base_type::accept(v);
                            ^
In file included from C:\dev\boost_1_59_0/boost/bind/bind.hpp:1163:0,
                 from C:\dev\boost_1_59_0/boost/bind.hpp:22,
                 from C:\dev\boost_1_59_0/boost/thread/detail/thread.hpp:30,
                 from C:\dev\boost_1_59_0/boost/thread/thread_only.hpp:22,
                 from C:\dev\boost_1_59_0/boost/thread/thread.hpp:12,
                 from C:\dev\boost_1_59_0/boost/thread.hpp:13,
                 from src/libslic3r/libslic3r.h:10,
                 from src/xsinit.h:53,
                 from src\libslic3r\utils.cpp:1:
C:\dev\boost_1_59_0/boost/bind/bind_template.hpp:326:40: error: macro "PerlSock_accept" requires 3 arguments, but only 1 given
     template<class V> void accept(V & v) const
                                        ^
C:\dev\boost_1_59_0/boost/bind/bind_template.hpp:334:20: error: macro "PerlSock_accept" requires 3 arguments, but only 1 given
         l_.accept(v);
                    ^
In file included from C:\dev\boost_1_59_0/boost/bind.hpp:22:0,
                 from C:\dev\boost_1_59_0/boost/thread/detail/thread.hpp:30,
                 from C:\dev\boost_1_59_0/boost/thread/thread_only.hpp:22,
                 from C:\dev\boost_1_59_0/boost/thread/thread.hpp:12,
                 from C:\dev\boost_1_59_0/boost/thread.hpp:13,
                 from src/libslic3r/libslic3r.h:10,
                 from src/xsinit.h:53,
                 from src\libslic3r\utils.cpp:1:
C:\dev\boost_1_59_0/boost/bind/bind.hpp:1538:17: error: macro "PerlSock_accept" requires 3 arguments, but only 1 given
     t.accept( v );
                 ^
In file included from C:/Strawberry/c/x86_64-w64-mingw32/include/c++/locale:41:0,
                 from C:\dev\boost_1_59_0/boost/io/ios_state.hpp:17,
                 from C:\dev\boost_1_59_0/boost/thread/detail/thread.hpp:35,
                 from C:\dev\boost_1_59_0/boost/thread/thread_only.hpp:22,
                 from C:\dev\boost_1_59_0/boost/thread/thread.hpp:12,
                 from C:\dev\boost_1_59_0/boost/thread.hpp:13,
[... 149 lines cut ...]
    ^
C:\STRAWB~1\perl\lib\CORE/XSUB.h:647:20: note: in expansion of macro 'PerlProc_sleep'
 #    define sleep  PerlProc_sleep
                    ^
C:\dev\boost_1_59_0/boost/thread/win32/thread_primitives.hpp:415:25: note: in expansion of macro 'sleep'
             inline void sleep(unsigned long milliseconds)
                         ^
In file included from C:\dev\boost_1_59_0/boost/tuple/tuple.hpp:28:0,
                 from C:\dev\boost_1_59_0/boost/thread/csbl/tuple.hpp:15,
                 from C:\dev\boost_1_59_0/boost/thread/detail/invoker.hpp:32,
                 from C:\dev\boost_1_59_0/boost/thread/future.hpp:20,
                 from C:\dev\boost_1_59_0/boost/thread.hpp:24,
                 from src/libslic3r/libslic3r.h:10,
                 from src/xsinit.h:53,
                 from src\libslic3r\utils.cpp:1:
C:\dev\boost_1_59_0/boost/tuple/detail/tuple_basic.hpp:45:28: error: expected '}' before end of line
 #pragma GCC diagnostic push
                            ^
[... cut ...]
C:\dev\boost_1_59_0/boost/tuple/detail/tuple_basic.hpp:45:28: error: expected '}' before end of line
C:\dev\boost_1_59_0/boost/tuple/detail/tuple_basic.hpp:45:28: error: expected declaration before end of line
error building xs.dll file from 'src\libslic3r\utils.cpp' at C:/Strawberry/perl/lib/ExtUtils/CBuilder/Platform/Windows.pm line 129.
FAIL
! Installing ./xs failed. See C:\Users\Martin\.cpanm\work\1480216552.10848\build.log for details. Retry with --force to force install it.
The XS/C++ code failed to compile, aborting

lordofhyphens, 2016-11-27:
> template void accept(V & v) const
> open(const basic_string&, const locale&, const char*) const;
> void wait(unique_lock& m,predicate_type pred)

You found more Perl macros that probably need undefining.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2016-11-27:
Go to xs/src/xsinit.h and add an #undef line for each of those PerlProc_* macros.

So:

`PerlProc_sleep` becomes `#undef sleep`

and so on. Continue doing so until it builds or another kind of error not mentioning macro expansions happens.

Tinchus2009, 2016-11-29:
Somehow that solution doesnt work. I have added a couple of #undef lines, but most of the named macros that appears in the errors, are already undef in the xsinit.h file. I have uploaded the last error (the log is larger than that but my cmd screen just scrolls more than I can copy.)
[error_log.txt](https://github.com/alexrj/Slic3r/files/618083/error_log.txt)

Tinchus2009, 2016-11-30:
Problems still exists. @lordofhyphens  instructed me to use perl 32 bit. I did that, and still doesnt build, the error are atached. Also, I tried perl 5.18 64 bits, I have atacjed the log with errors, they look the same as with the 32 bit version
[log_perl32bit.txt](https://github.com/alexrj/Slic3r/files/621019/log_perl32bit.txt)
[log_perl5.18_64bit.txt](https://github.com/alexrj/Slic3r/files/621021/log_perl5.18_64bit.txt)

lordofhyphens, 2016-11-30:
Start by cleaning your build directory (run perl Build clean in the xs
directory), or delete the entire Slic3r dir and check it out again.

I've never seen that error before on my Win10 system or on the Win8 build
server.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2016-11-30:
I am trying to reproduce on a clean 5.18 64-bit install of Strawberry perl.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

Tinchus2009, 2016-11-30:
ok, I have tried the  BUILD CLEAN iside XS directory: the same errors . I tried a fresh build from scratch, and the same build errors reported

lordofhyphens, 2016-11-30:
I have no idea what is going on with your system/environment. I can't
reproduce this on my windows 10 system.

You may wish trying to build the prusa3d fork. If you are getting the same
issues it is probably something in your local environment.

Tinchus2009, 2016-11-30:
Ok...  But if the problem would be my system/enviroment, why the build was
ok till last aditions?....
And yes, the prusad built finished ok last night, so,  without being a
coder, looks like the problem is here.

lordofhyphens, 2016-11-30:
The hang was from a test that was added recently from the Prusa3d fork. 
Something you could do is rename C:\Strawberry to C:\Strawberry.old, make a new C:\Strawberry and extract http://strawberryperl.com/download/5.18.2.2/strawberry-perl-5.18.2.2-64bit.zip into that dir. 

You will want to rebuild Boost as per the instructions. 
From there, do a clean checkout of current slic3r master and run both
`perl Build.PL` and `perl Build.PL --gui` and wait for things to sort themselves out. 

I just did this process last night on my Windows 10 laptop.

Tinchus2009, 2016-11-30:
Ok, I followed your last instructions. Now I get to this point:

t\01_trianglemesh.t ............. ok
t\03_point.t .................... ok
t\04_expolygon.t ................ ok
t\05_surface.t .................. ok
t\06_polygon.t .................. ok
t\07_extrusionpath.t ............ ok
t\08_extrusionloop.t ............ ok
t\09_polyline.t ................. ok
t\10_line.t ..................... ok
t\11_clipper.t .................. ok
t\12_extrusionpathcollection.t .. ok
t\13_polylinecollection.t ....... ok
t\14_geometry.t ................. ok
t\15_config.t ................... ok
t\16_flow.t ..................... ok
t\17_boundingbox.t .............. ok
t\18_motionplanner.t ............ ok
t\19_model.t .................... ok
t\20_print.t .................... ok
t\21_gcode.t .................... ok
t\22_exception.t ................ ok
t\23_config.t ................... ok
All tests successful.
Files=22, Tests=495,  3 wallclock secs ( 0.16 usr +  0.05 sys = 0.20 CPU)
Result: PASS
Building Slic3r-XS
Files found in blib\arch: installing files in blib\lib into architecture 
dependent library tree
Installing C:\Strawberry\perl\site\lib\auto\Slic3r\XS\XS.bs
Installing C:\Strawberry\perl\site\lib\auto\Slic3r\XS\XS.dll
Installing C:\Strawberry\perl\site\lib\Slic3r\XS.pm
OK
Successfully installed Slic3r-XS-0.01
2 distributions installed
The following prerequisites failed to install: IO::Scalar

lordofhyphens, 2016-11-30:
> perl-5.18.2.2-64bit.zip

It looks to me like you can ignore the IO::Scalar failure; the program at
least looks like it starts without it, and IO::Scalar can probably be made
to install separately with cpanminus (which is on my todo list).

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

Tinchus2009, 2016-11-30:
But the build stops there. It doesnt finish. So i can yry to install it
separatly with cpan?

lordofhyphens, 2016-11-30:
Yes. The important part of the build process is getting Slic3r-XS
installed, which happened.

Rest of it is getting other perquisites sorted out.

Tinchus2009, 2016-12-01:
Ok, something happened. I dont know how to install this IO::scalar.

I got tired and issued a perl build.pl --gui command anyway.... it 
builted with no problems. and slic3r started ok . I tried some slices 
and everything seems to work.

Couple of questions: if the  perl buul.pl command stopped, but then the 
one with the --gui completed, is there any functionality that will be 
missing?

The new aditions, is there any list I can check so I can test those new 
features/fixes?

Dont know why but I was thinking the "slice now" buttom would be 
implemented but I dont see it.

Minimum layer time was fixed too I guess?

lordofhyphens, 2016-12-01:
Look at the git log. You're basically drinking from the firehose here so
you should probably get cozy with the commit log to see what has been done
so far.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2016-12-04:
@Tinchus2009 give this a try https://bintray.com/lordofhyphens/Slic3r/download_file?file_path=slic3r-master-fde6e23.zip

Tinchus2009, 2016-12-04:
We crossed emails. I was exactly now writng this: I followed your 
instructions, despite slic3r seemed to have been builted and installed, 
a dependency failed: IO::scalar

I found this belongs to IO::stringy, wich is the one that fails to 
install when building slicer. Installing this amnually through cpan also 
fails, here it is the log:

cpanm (App::cpanminus) 1.7042 on perl 5.018002 built for 
MSWin32-x64-multi-thread
Work directory is C:\Users\Martin/.cpanm/work/1480740041.8564
You have make C:\Strawberry\c\bin\dmake.exe
You have LWP 6.15
Falling back to Archive::Tar 1.96
Searching IO::Scalar (0) on cpanmetadb ...
--> Working on IO::Scalar
Fetching http://www.cpan.org/authors/id/D/DS/DSKOLL/IO-stringy-2.111.tar.gz
-> OK
Unpacking IO-stringy-2.111.tar.gz
Entering IO-stringy-2.111
Checking configure dependencies from META.yml
Checking if you have ExtUtils::MakeMaker 6.58 ... Yes (7.24)
Configuring IO-stringy-2.111
Running Makefile.PL
Checking if your kit is complete...
Looks good
Generating a dmake-style Makefile
Writing Makefile for IO::Stringy
Writing MYMETA.yml and MYMETA.json
-> OK
Checking dependencies from MYMETA.json ...
Checking if you have ExtUtils::MakeMaker 0 ... Yes (7.24)
Building and testing IO-stringy-2.111
cp lib/IO/AtomicFile.pm blib\lib\IO\AtomicFile.pm
cp lib/IO/ScalarArray.pm blib\lib\IO\ScalarArray.pm
cp lib/IO/Lines.pm blib\lib\IO\Lines.pm
cp lib/IO/Wrap.pm blib\lib\IO\Wrap.pm
cp lib/IO/InnerFile.pm blib\lib\IO\InnerFile.pm
cp lib/IO/Stringy.pm blib\lib\IO\Stringy.pm
cp lib/IO/Scalar.pm blib\lib\IO\Scalar.pm
cp lib/IO/WrapTie.pm blib\lib\IO\WrapTie.pm
"C:\Strawberry\perl\bin\perl.exe" "-MExtUtils::Command::MM" 
"-MTest::Harness" "-e" "undef *Test::Harness::Switches; test_harness(0, 
'blib\lib', 'blib\arch')" t/*.t
t/IO_InnerFile.t ....
Failed 7/7 subtests
t/IO_Lines.t ........ ok
t/IO_Scalar.t ....... ok
t/IO_ScalarArray.t .. ok
t/IO_WrapTie.t ...... ok
t/simple.t .......... ok
t/two.t ............. ok

Test Summary Report
-------------------
t/IO_InnerFile.t  (Wstat: 0 Tests: 7 Failed: 7)
   Failed tests:  1-7
Files=7, Tests=125,  0 wallclock secs ( 0.01 usr +  0.06 sys = 0.08 CPU)
Result: FAIL
Failed 1/7 test programs. 7/125 subtests failed.
dmake.exe:  Error code 255, while making 'test_dynamic'
-> FAIL Installing IO::Scalar failed. See 
C:\Users\Martin\.cpanm\work\1480740041.8564\build.log for details. Retry 
with --force to force install it.

I forced the install, and it went ok. After that, building slic3r was 
completly succesful.

lordofhyphens, 2017-01-26:
@Tinchus2009 Are you still having issues building, or can this be closed?

Tinchus2009, 2017-01-26:
Sorry, I thought this was closed, My last reports was that foceing the install solved the issue. But latest dev version didnt show this problem at all. Closeing

PR #3593 (2016-11-30): Fix perl redefining multiple functions used by boost
Should fix remaining compile issues on Windows.

Fixes #3583
closed 2017-01-26 by Tinchus2009