slic3r/Slic3r #3584: Feature request: extra perimeters for interior openings (holes)

JBKingdon, 2016-11-24:
### Version

next

### Operating system type + version

any/all

### Behavior

When making holes for self-tapping screws the wall of the hole needs to be fairly thick to take the thread and provide adequate support for the fixing. I sometimes set 3 or 4 perimeters because I'm using threaded holes, but really only need 2 perimeters for the outside walls of the part.

Would it be possible to detect an 'interior' perimeter (I realize that is a non-trivial problem during slicing) and apply a configurable number of extra perimeters to it? Even a partial implementation that only worked on circles in the slicing plane would be very useful as that covers (or could be made to cover) 90% of my use cases.

_Is this a new feature request?_ yes

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

N/A

lordofhyphens, 2016-11-24:
This should be doable already. with modifier meshes if I remember
correctly.

JBKingdon, 2016-11-24:
Yes, agreed, modifiers would handle this. However modifiers are quite hard to use - you have to go right back to cad to generate the modifier volumes - so a way of handling this at slice time would I think have additional value. I accept that "a slightly easier way" of doing something is not a strong case though.

lordofhyphens, 2016-11-24:
I agree that adding UI capability to define modifier meshes would be a good
thing.

JBKingdon, 2016-11-24:
Ah, now that would be a more generally applicable and useful approach! Sounds like a lot of work though?

lordofhyphens, 2016-11-27:
Well, I just implemented a version of that, so it was quite a bit of work. 
See #3590