slic3r/Slic3r #3585: Cubic Infill at bigger layer heights wrong

callimero, 2016-11-24:
### Version
Slic3r 1.31.6 - Prusa Edition -win64

### Operating system type + version
win 8.1 64bit

### Behavior
The very nice cubic infill seems to be wrongly generated at higher layer heigths.
[image]

left layer heigth 0.24mm right same scene, same params, only layer heigth 0.48mm

Looks for me as if the offset from layer to layer is not calculated correctly (to big, so the upper layer is not supported from the lower layer)

I think this was also in the prusa forum but I could not find it.

bubnikv, 2016-11-24:
Well, you have to keep the infill line sufficiently wide for the sloping infill walls to be buildable. On this particular picture the infill lines are too thin to overlap sufficiently at the 45 degree slope of the infill walls.

bubnikv, 2016-11-24:
By the way, layer height 0.48 is not printable with 0.45 infill line width.

callimero, 2016-11-24:
I think I put it in the wrong slic3r repo.... 

However please slic3r-program please care for such user too stupid errors.... ;-)

callimero, 2016-11-24:
I am closing I guess the cubic infill is only in the prusa fork

closed 2016-11-24 by callimero