slic3r/Slic3r #3589: Division by zero when chaning to spiralize mode

jtpedersen, 2016-11-26:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_ dfd2466a724acf12a0e48d0645f381b01d45680a

### Operating system type + version
_What OS are you using, and state any version #s_ Ubuntu

### Behavior
* _Describe the problem_
Illegal division by zero at .Slic3r/lib/Slic3r/GUI/3DScene.pm line 373.

* _Steps needed to reproduce the problem_
Enable spiralize mode
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
no crash
* _Actual Results_
crasch
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_ no

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

jtpedersen, 2016-11-26:
i can work around it by setting the $max_size to  max($max_size, 1). But it looks really weird when starting up....

kino3d, 2016-11-27:
I have the same error, if you want you can just comment out the line 373 in 3Dscene.pm 
#    $self->_zoom($min_viewport_size / $max_size);

i will not zoom every time the object is recalculated which is better imho :-) 
+1 to become an option

jreinam, 2016-11-27:
Or you can test for zero before updating zoom, so if zero don't update zoom/camera/viewport.

```

diff --git a/lib/Slic3r/GUI/3DScene.pm b/lib/Slic3r/GUI/3DScene.pm
index 61d45f1..141ecf5 100644
--- a/lib/Slic3r/GUI/3DScene.pm
+++ b/lib/Slic3r/GUI/3DScene.pm
@@ -370,12 +370,14 @@ sub zoom_to_bounding_box {
     # bounding box
     my $max_size = max(@{$bb->size}) * 2;
     my $min_viewport_size = min($self->GetSizeWH);
+       if ($max_size!=0) {
     $self->_zoom($min_viewport_size / $max_size);
     
     # center view around bounding box center
     $self->_camera_target($bb->center);
     
     $self->on_viewport_changed->() if $self->on_viewport_changed;
+       }
 }
 
 sub zoom_to_bed {

```

lordofhyphens, 2016-11-27:
I can't get this to reproduce in master branch on my system. Wahoo. Probably good though to use that safeguard as defensive coding though.

alexrj, 2016-11-27:
I fixed this in a7693c4719c0e0e0129aea7461706d2ed53a52cc

jreinam, 2016-11-27:
Same issue reproduces here:

GUI crashes because division by zero.

After starting program and loading an .stl file (from "Add..." tab)
If swap tab from 3D to "Preview" when status bar says "Processing triangular mesh..."
then illegal division by zero

$ ./slic3r.pl 
Illegal division by zero at ***/Slic3r/lib/Slic3r/GUI/3DScene.pm line 373.
Perl exited with active threads:
	1 running and unjoined
	0 finished and unjoined
	0 running and detached

lordofhyphens, 2016-11-27:
@jreinam with a fresh checkout?

jreinam, 2016-11-27:
Tested with fresh checkout and gives the issue.

Use a medium-big stl file if want to reproduce, as you need to do at the time of "Processing triangular mesh..."
Also if you click to "Delete" or "Delete All" at that time.

I think the process "triangular mesh" maybe is in background process and related to the variable initialization/update, but I don't know the program internals.  Nevertheless the proposed solution solves that as workaround.

gege2b, 2016-11-28:
I got the same problem with 9eb28371def188032242fbfbaa56475e0440efb5

```
Illegal division by zero at /home/gege/3Dprint/Slic3r/lib/Slic3r/GUI/3DScene.pm line 379.
Perl exited with active threads:
        1 running and unjoined
        0 finished and unjoined
        0 running and detached
```
Slicing seems to be stuck on "processing triangulated mesh" (but maybe this is due to the [stl](http://www.thingiverse.com/download:1324091) I'm using?)

jreinam, 2016-11-29:
Solved also if uncomment that two lines.
It also adds to scene the volume with transparency. If not want to show then change to color->[3] = 0.0 .

```
diff --git a/lib/Slic3r/GUI/Plater/3DPreview.pm b/lib/Slic3r/GUI/Plater/3DPreview.pm
index fa71fcb..68b8f8c 100644
--- a/lib/Slic3r/GUI/Plater/3DPreview.pm
+++ b/lib/Slic3r/GUI/Plater/3DPreview.pm
@@ -123,8 +123,8 @@ sub load_print {
         foreach my $object (@{$self->print->objects}) {
             $self->canvas->load_print_object_toolpaths($object);
             
-            #my @volume_ids = $self->canvas->load_object($object->model_object);
-            #$self->canvas->volumes->[$_]->color->[3] = 0.2 for @volume_ids;
+            my @volume_ids = $self->canvas->load_object($object->model_object);
+            $self->canvas->volumes->[$_]->color->[3] = 0.2 for @volume_ids;
         }
         $self->canvas->zoom_to_volumes;
         $self->_loaded(1);
```

jreinam, 2016-11-29:
As conclusion I have the question is whether it is really needed to update the preview there or not. There are no changes to the behavior if it is not updated¿?. Maybe we only need to comment the "zoom update" line:
```
diff --git a/lib/Slic3r/GUI/Plater/3DPreview.pm b/lib/Slic3r/GUI/Plater/3DPreview.pm
index fa71fcb..60a0841 100644
--- a/lib/Slic3r/GUI/Plater/3DPreview.pm
+++ b/lib/Slic3r/GUI/Plater/3DPreview.pm
@@ -126,7 +126,7 @@ sub load_print {
             #my @volume_ids = $self->canvas->load_object($object->model_object);
             #$self->canvas->volumes->[$_]->color->[3] = 0.2 for @volume_ids;
         }
-        $self->canvas->zoom_to_volumes;
+        #$self->canvas->zoom_to_volumes;
         $self->_loaded(1);
     }
     
```

jtpedersen, 2016-11-30:
it would still probably be a good idea to guard agains the 0 division. perhaps just return if $max_size is 0. ?

jreinam, 2016-11-30:
I think so, using one of the 3 above alternative solutions only traces/solves when it's called without the bounding box. To clean the code.

Testing a variable for 0 is almost no computational cost and avoids the unexpected annoying crashes.

jreinam, 2016-12-02:
The core of the problem is that we don't draw anything before STEP_PERIMETERS, or STEP_INFILL if there isn't any perimeters on config. As no draw anything also there isn't any bounding_box.

As draw anything is no real problem go ahead with merging [https://github.com/alexrj/Slic3r/issues/3591](https://github.com/alexrj/Slic3r/issues/3591)

PR #3591 (2016-12-16): Avoid recentering viewport if the bounding box is max size 0
Defensive coding - fix provided by @jreinam

Fixes #3589
closed 2016-12-16 by alexrj