slic3r/Slic3r #3592: apt-get build-dep slic3r error on raspbian

linzhiqi, 2016-11-28:
### Operating system type + version
_What OS are you using, and state any version #3415 
```
RASPBIAN JESSIE LITE
Minimal image based on Debian Jessie
Version:September 2016
Release date:2016-09-23
Kernel version:4.4
```
### Behavior
* when install dependencies, got error
````
$ sudo apt-get build-dep slic3r
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages have unmet dependencies:
 libwx-perl : Depends: libwxgtk-media3.0-0 (>= 3.0.2) but it is not going to be installed
              Depends: libwxgtk3.0-0 (>= 3.0.2) but it is not going to be installed
              Depends: wxperl-gtk2-3-0-2-uni-gcc-3-4
              Depends: libalien-wxwidgets-perl (>= 0.65+dfsg~) but it is not going to be installed
E: Build-dependencies for slic3r could not be satisfied.
```
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
Not

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

lordofhyphens, 2016-11-28:
This is a question for the official Debian package maintainer, not a Slic3r bug. 

https://packages.debian.org/search?keywords=slic3r

closed 2016-11-28 by lordofhyphens

linzhiqi, 2016-11-28:
@lordofhyphens does it mean, slic3r can not be built from source on Debian Jessie for now?

bubnikv, 2016-11-28:
probably yes, but from source, not from the debian source package.

linzhiqi, 2016-11-28:
@bubnikv do you mean the debian branch in https://github.com/lordofhyphens/Slic3r should be ok to be built from source?

lordofhyphens, 2016-11-28:
That source is pretty old; I haven't updated it in a while (I use
local::lib to not wreck my debian perl).

I will see what I can do to fix it, but that package isn't official (nor am
I the Debian maintainer).

lordofhyphens, 2016-11-28:
As I said earlier, you should contact the official maintainer of the Debian Slic3r package (or Raspberrian distro) regarding apt-get build-deps issues.