slic3r/Slic3r #3595: Compilation failure with Boost 1.54

jabdoa2, 2016-11-30:
### Version
_Version of Slic3r used goes here_
master

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_
1.2.9-485-g3f3f696

### Operating system type + version
_What OS are you using, and state any version #s_
Ubuntu 14.04 x64

### Behavior
Compilation fails:

```
perl Build.PL

[...]

cc -I/usr/lib/perl/5.18/CORE -fPIC -xc++ -D_FILE_OFFSET_BITS=64 -Isrc/libslic3r -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DBOOST_LIBS -DNDEBUG -Isrc -Ibuildtmp -c -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fstack-protector -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g -o src/admesh/util.o src/admesh/util.c
In file included from src/admesh/util.c:28:0:
src/admesh/stl.h:29:47: fatal error: boost/predef/detail/endian_compat.h: File not found
 #include <boost/predef/detail/endian_compat.h>
                                               ^
compilation terminated.
error building src/admesh/util.o from 'src/admesh/util.c' at /usr/share/perl/5.18/ExtUtils/CBuilder/Base.pm line 177.
FAIL
! Installing ./xs failed. See .cpanm/work/1480499466.28998/build.log for details. Retry with --force to force install it
```

Regression caused by this commit: https://github.com/alexrj/Slic3r/commit/d88f05945d40e82fc887f9130f716355d534c690

boost/predef/detail/endian_compat.h does not exist in Boost 1.54 shipped with Ubuntu 14.04. Slic3r works fine commenting out those lines.

alexrj, 2016-11-30:
Thank you, we need macros for excluding those lines with old Boost versions.

alexrj, 2016-11-30:
Actually, the file was not there but I think it's not supposed to be included directly. I pushed a fix now. Let me know if it works for you.

jabdoa2, 2016-11-30:
works. thanks

closed 2016-11-30 by jabdoa2