slic3r/Slic3r #3597: Spurious retractions

jabdoa2, 2016-11-30:
### Version
_Version of Slic3r used goes here_
master

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_
1.2.9-485-g3f3f696

### Operating system type + version
_What OS are you using, and state any version #s_
Ubuntu 14.04 x64

### Behavior
* _Describe the problem_
Printer retracts during a lot print. Retraction is 1.5mm which works but sometimes when printing infill after diameters it retracts 2cm or more which breaks the print.
* _Steps needed to reproduce the problem_
Slice skull of http://www.thingiverse.com/make:270358
* _Expected Results_
No spurious retraction. 
* _Actual Results_
Retracts unexpectedly.

```
M106 S178.5
G1 Z1.550 F9600.000
G1 E1.91193 F1800.00000
G92 E0
G1 Z1.850 F9600.000
G1 X157.643 Y191.570 F9600.000
G1 Z1.550 F9600.000
G1 E1.50000 F1800.00000
G1 F3900
G1 X154.127 Y190.026 E1.71062
G1 X152.751 Y189.325 E1.79534
G1 X152.189 Y188.984 E1.83142
G1 X151.810 Y188.725 E1.85658
G1 X151.445 Y188.420 E1.88269
G1 X151.139 Y188.136 E1.90560
G1 X150.925 Y187.877 E1.92403
G1 X150.745 Y187.613 E1.94156
G1 X150.596 Y187.305 E1.96032
G1 X150.512 Y187.012 E1.97706
G1 X150.475 Y186.736 E1.99232
G1 X150.473 Y186.470 E2.00692
G1 X150.555 Y185.258 E2.07355
G1 X150.781 Y184.910 E2.09628
G1 X151.032 Y184.586 E2.11878
G1 X151.623 Y184.016 E2.16383
G1 X152.020 Y183.465 E2.20108
G1 X152.121 Y183.286 E2.21235
G1 X152.518 Y182.805 E2.24655
G1 X152.629 Y182.639 E2.25751
G1 X152.944 Y182.661 E2.27479
G1 X153.634 Y182.784 E2.31329
G1 X154.043 Y182.888 E2.33641
G1 X154.422 Y183.016 E2.35838
G1 X154.851 Y183.204 E2.38407
G1 X155.675 Y183.615 E2.43458
G1 X156.216 Y183.931 E2.46895
G1 X158.939 Y185.602 E2.64420
G1 X160.964 Y186.736 E2.77151
G1 X161.865 Y187.201 E2.82716
G1 X164.389 Y188.343 E2.97910
G1 X167.592 Y189.733 E3.17067
G1 X169.599 Y190.540 E3.28933
G1 X170.582 Y190.884 E3.34644
G1 X171.737 Y191.260 E3.41307
G1 X173.234 Y191.691 E3.49854
G1 X176.210 Y192.455 E3.66709
G1 X179.976 Y193.330 E3.87921
G1 X184.893 Y194.300 E4.15410
[... 546 lines cut ...]
; fill_density = 20%
; fill_pattern = honeycomb
; gap_fill_speed = 20
; infill_every_layers = 1
; infill_extruder = 1
; infill_extrusion_width = 0
; infill_overlap = 15%
; infill_speed = 80
; overhangs = 1
; perimeter_extruder = 1
; perimeter_extrusion_width = 0
; perimeter_speed = 65
; perimeters = 3
; small_perimeter_speed = 55
; solid_infill_below_area = 70
; solid_infill_every_layers = 0
; solid_infill_extruder = 1
; solid_infill_extrusion_width = 0
; solid_infill_speed = 70
; thin_walls = 1
; top_infill_extrusion_width = 0
; top_solid_infill_speed = 15
; top_solid_layers = 3

```

lordofhyphens, 2016-11-30:
In the future, please use File->Export Config to get a config.ini file and zip that and any attachments and attach that. 

Dumping gcode in plain text like that is painful to deal with.

lordofhyphens, 2016-11-30:
It would probably help a lot if you turned on firmware retraction to highlight the retract/restore commands (or turned on verbose gcode which should show them). 

Slic3r is going to retract at the end of every loop when it has to cross perimeters (as noted in your gcode). 

I've attached a slice from a build of 3700950 on windows. 
[config-debug.zip](https://github.com/alexrj/Slic3r/files/622450/config-debug.zip)

jabdoa2, 2016-11-30:
Ok will do next time. Noteworthy in the paste was  Filament used: -5861.5mm. In the snippet it would retract 35mm which removed all Filament from the extruder. Same model works with same settings in stable.

Tinchus2009, 2016-12-07:
Im observing a what I think is the same behaviour: I have atacched a stl and my configs (slic3r version is the last version I got from git pull command). My retraction setting are 0.5 mm. but like soemthign around Z + 30 mm the extruder starts makeing HUGE retractions, like 3 cms or more I would say, that completly ruins the print since after this retraction, the filament is only pulled in 0.5 mm,,,
[pruebas_impresion.zip](https://github.com/alexrj/Slic3r/files/636857/pruebas_impresion.zip)

lordofhyphens, 2016-12-07:
@Tinchus2009 turn off pressure advance.

Tinchus2009, 2016-12-07:
I dont use that feature, in the printer settings you mean? there, pressure advance is set to 0

Tinchus2009, 2016-12-07:
I have repeated the print: same object, same config, only difference was that I change position of the object. It happened again, but this time I decided to let the print continue. The issue arised somewhere between Z = 40 and 47 mm. A huge retraction hapened there, after that the print conued with no issues so far (still printing now). I generated the verbose gcode file (atached)
[Assem1_-_Part8Assem1-1.zip](https://github.com/alexrj/Slic3r/files/637080/Assem1_-_Part8Assem1-1.zip)

jabdoa2, 2016-12-07:
Same issue for me. It starts fine and after a few layers it retracts large amounts of filament. Same model with same settings works in stable. Also the total filament used is negative in dev but not in stable.

lordofhyphens, 2016-12-07:
Gap fill may be the culprit then. Turn on verbose gcode output and it will
tell you in the gcode when it is retract.

Tinchus2009, 2016-12-08:
I dont use autospeed nor pressure advance

lordofhyphens, 2016-12-08:
As I said, try disabling gap fill and turn on verbose gcode to see what the
long negative moves look like in the printing process. They probably aren't
actually retract moves.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2016-12-08:
@alexrj any idea offhand why there would be negative E values in the extrusion? Verbose gcode is tagging them as perimeters. It's definitely not a retract though, as there's X/Y moves along with it.

lordofhyphens, 2016-12-08:
Looks like a regression of some kind since 1.2.9 at least. Only have all the commits of course to check.

lordofhyphens, 2016-12-08:
Still in the middle of the bisect. 893fc2343f0aa3855078f20561d2f7a977610791 exhibits the problem.

lordofhyphens, 2016-12-08:
Accordng to git bisect, the problem commit is 24afffc421ebd8a9297528bdfa038ba33490e3c2 

Tomorrow I'll write up a regression test with a minimum case of the trex jaw (odds are there's negative flow in some of these areas for some reason). 

@bubnikv Unless you've done a ton more work since this commit this'll be in the Prusa build as well.

lordofhyphens, 2016-12-08:
The observed behavior is perimeters are being drawn with negative feedrate, unknown at present why.

bubnikv, 2016-12-08:
> Accordng to git bisect, the problem commit is 24afffc
<https://github.com/alexrj/Slic3r/commit/24afffc421ebd8a9297528bdfa038ba33490e3c2>

> Tomorrow I'll write up a regression test with a minimum case of the trex
jaw (odds are there's negative flow in some of these areas for some reason).

> @bubnikv <https://github.com/bubnikv> Unless you've done a ton more work
since this commit this'll be in the Prusa build as well.

I am not quite sure whether the bisect is a right tool here as the file has
been touched later on to fix some issues.

https://github.com/prusa3d/Slic3r/blame/master/xs/src/libslic3r/ExtrusionEntity.cpp

@Tinchus2009

Would you please test the latest prusa3d build for this issue? Thanks

https://github.com/prusa3d/Slic3r/releases/tag/version_1.31.6

lordofhyphens, 2016-12-08:
@bubnikv I ran the bisect with an initial good and bad of 1.2.9 and HEAD,
skipping commits that failed to build and verifying the output after every
successful pass. I am fairly confident that the regression was introduced
there (binary search), but I intend to build a regression test anyway for
future safety.

Obviously if you changed the file after that commit then it may not be
reflected in your fork.

lordofhyphens, 2016-12-09:
@bubnikv Looks like you found the issue in the prusa3d fork and fixed it already, the original commit used the ExtrusionPath constructor that only supplied the role and used default args for everything else.

You later changed it to call the constructor to give the two new paths the same flow amount and width/height. This differs from the original (prior to that commit) in that @alexrj used what looks like the implicit copy constructor (via =). 

Here's a slice of the trex object that shows the problem, currently working on a regression test using it.
[3597-failure.zip](https://github.com/alexrj/Slic3r/files/640944/3597-failure.zip)

closed 2016-12-09 by alexrj

Tinchus2009, 2016-12-09:
Sorry to be kate. I have tested the prusa version you told me, useing the same stl, it looks like the problem is not present. Either way, I can see you solve it here too, great!

PR #3609 (2016-12-09): Fixed small issue with perimeter path splitting could cause negative …
…flow/feedrates.

Fixes #3597