slic3r/Slic3r #3598: Changing some settings doesn't work, uses old/wrong values

VanessaE, 2016-11-30:
### Version

Built from sources, git commit 11585b88d5a1063f28a17ae5a90698e2c23ddcda

### Operating system type + version

Debian 8.6

### Behavior

When you change a setting, Slic3r often does not do a full re-slice to account for that setting-change when you click `Export G-code`, but the bad part here is that it sometimes *does* do a full re-slice, but the setting you changed is not applied.  Rather, the previous value is used, or so it appears.

For example, load a model such as http://www.thingiverse.com/thing:1363023 and go through the usual steps as if you are going to print it, and turn on pressure advance, K=0.5.  Slice it and examine the result in a g-code analyzer.  Change the value, re-slice, and load the g-code into your analyzer (in a separate instance).  Repeat this a few more times.

Do that enough times and eventually the last value you sliced with will not be what ends up in the g-code.

#### STL/Config (.ZIP) where problem occurs

[Slic3r_config_bundle_1.3.0-dev-2016-11-30.ini.zip](https://github.com/alexrj/Slic3r/files/622097/Slic3r_config_bundle_1.3.0-dev-2016-11-30.ini.zip)

Note that I have background processing *disabled*.  This config file shouldn't actually be needed, as the problem seems to be independent of the chosen Print, Filament, or Printer profile.  Nevertheless, I was last using the "60 mm/sec 0.2mm layers" Print profile therein, and the "Royal Blue" filament profile.

bubnikv, 2016-11-30:
Just a note to the reaction on change of configuration parameters. A change
of a configuration parameter invalidates all steps starting with a step,
which is explicitly hard coded in the software. What I want to say is, your
bug may or may not be specific to the pressure advance parameter.

VanessaE, 2016-11-30:
I didn't figure it was specific to Pressure Advance, that's just the setting I happened to be fiddling with when I realized the issue.

bubnikv, 2016-11-30:
As I said, it may or may not be specific to that particular parameter.

lordofhyphens, 2016-12-02:
@VanessaE Were you able to reproduce with a parameter that isn't with Pressure Advance?

VanessaE, 2016-12-02:
I have not checked if other settings are affected.

lordofhyphens, 2016-12-02:
Please do. Pressure regulator happens late and I think modifies the actual
flow (you would have to check the PressureRegulator.pm to be sure). If it
does that then the priors for the function wouldn't be correct.

alexrj, 2016-12-03:
Changing the Pressure Advance option is not supposed to trigger a reslice because that option only affects G-code export.

So, if the issue you're reporting is: Slic3r does not trigger a reslice after changing Pressure Advance, that's not a bug and it's correct behavior.
If the issue you're reporting is: changes to the Pressure Advance option are not reflected in the exported G-code, we need more evidence in order to reproduce the problem. I just tried changing it and exporting G-code several times, and the output was always consistent with the option.

VanessaE, 2016-12-03:
Odd.  All I know for sure is I turned the value up over the course of a few prints, then turned it back down (way down), and the print looked as though it was still at the higher value.  Maybe I just forgot to reload the correct gcode.

closed 2016-12-03 by VanessaE