slic3r/Slic3r #3599: Can I prebuild my customized slic3r from source code?

wucm1994, 2016-12-01:
### Version
Hash value for the version:
f7885d19515aced7693494f19c6768eed54a9427

### Operating system type + version
Windows 10 64bit 1607

### Behavior
Hi! I'm planing to prebuild the slic3r from my customized source code (primarily for the purpose of rendering on multiple PCs in parallel without environment configuration), just like the release version you provided. Is there possible to do so? Thanks.

lordofhyphens, 2016-12-01:
If you're talking about the cava-packaged version, it's certainly possible but I don't have the tools or knowledge myself to do it (@alexrj would). Cava's Perl support is on the older side though so I'm not that 

The other way to do it is how the Prusa3d does their build. There's a version of the packaging script that I use at https://github.com/alexrj/Slic3r/blob/master/utils/package_win32.ps1. It uses PAR::Packer. 

That script is the distillation of the work written down in the wiki: https://github.com/alexrj/Slic3r/wiki/Building-Slic3r-on-OSX-and-Windows-for-Distribution

closed 2016-12-01 by lordofhyphens

wucm1994, 2016-12-01:
Thanks, I was very appreciated that.