slic3r/Slic3r #3600: Cannot launch because of undefined OpenGL AUTOLOAD subroutine

TheJJ, 2016-12-03:
### Version
1.2.9-503-gfde6e2e6
fde6e2e61deabd097a3bf565d4948077147b25ab

### Operating system type + version
Gentoo Linux

### Behavior

The launch doesn't work:

```
$ slic3r.pl --gui
Goto undefined subroutine &AutoLoader::AUTOLOAD at /usr/lib64/perl5/vendor_perl/5.24.1/x86_64-linux-thread-multi/OpenGL.pm line 6110.
```

Perl is up to date on my machine, I ran `perl-cleaner --all`. The only thing I found related to that error is https://stackoverflow.com/questions/24989072/first-time-in-perl-opengl.

If you have no idea whether `slic3r` itself caused that error, please provide me some stept I can try narrowing down the problem as I have no idea how to debug Perl.

lordofhyphens, 2016-12-03:
> questions/24989072/first-time-in-perl-opengl.

Weird. I have seen this on the Windows build server output (which I am
still debugging) but I have never seen this before on a Linux system  (I
use debian).

I will look at the 3d code and see how we are initializing it.

lordofhyphens, 2016-12-03:
Could you make sure that AutoLoader is installed via cpanm? 
`cpanm AutoLoader`

TheJJ, 2016-12-03:
I have installed autoloader via my package manager of course: `virtual/perl-AutoLoader`, which depends on `dev-lang/perl` (the main interpreter).
This package has the `AutoLoader.pm` installed:
```
qlist -e dev-lang/perl | ag autoloader
/usr/lib64/perl5/5.24.1/AutoLoader.pm
```
So i assume i have installed AutoLoader.

I have to say, slic3r worked before on my machine. Some update broke it (either your git or some perl stuff).

lordofhyphens, 2016-12-03:
Always bet on perl stuff, the only GL code that was changed was to avoid
having problems with changes in OpenGL.pm.

Something you could try is to downgrade OpenGL.pm. To do this, find the
line in Build.PL that references OpenGL and change the following 0 to <
0.70. Rerun --gui and see what happens.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

TheJJ, 2016-12-03:
I have [OpenGL.pm 0.670.400](https://packages.gentoo.org/packages/dev-perl/OpenGL) installed via portage. The git version I have is also installed [via portage](https://gitweb.gentoo.org/repo/gentoo.git/tree/media-gfx/slic3r/slic3r-9999.ebuild). But if your build.pl requires OpenGL.pm >= 0.70 then maybe my "outdated" 0.670 is the cause?

lordofhyphens, 2016-12-03:
It shouldn't require 0.70, but we did just have recent commit to work
around a segfault caused by something that changed in 0.70.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

TheJJ, 2016-12-04:
When i downgrade the slic3r via portage to the [latest packaged version 1.2.9-r2](https://gitweb.gentoo.org/repo/gentoo.git/tree/media-gfx/slic3r/slic3r-1.2.9-r2.ebuild), it works again. I did not downgrade any other package except slic3r.

lordofhyphens, 2016-12-04:
A lot has changed since then. I just wish I had a reproduction case as I
can't get it to break on my system in that way.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

alexxy, 2017-01-03:
Actualy have same problems here... 
Btw is perl-local libe realy used by slic3r?

lordofhyphens, 2017-01-03:
@Alexey it is, but only for dependencies that aren't in your system libraries  (in builds after mid December).

I haven't been able to reproduce this on any of my systems. You may wish to contact the gentoo package maintainer and see if that person has any insights.

alexxy, 2017-01-11:
Btw, I tryed slic3r from prusa3d fork. It doesnt have this issue, but it crashes if perl/opengl is 0.70. Earlyer version of this package works fine

lordofhyphens, 2017-01-11:
There is something going on with your local system or distro.

I don't think @bubnikv pulled my VBO change, which means that the prusa3d
fork wouldn't be compatible with 0.70.

alexxy, 2017-01-11:
Problem is that slic3r is only app that failing with perl autoload =/ may be it's related to perl version (5.24.x)
Anyway prusa3d fork work, and previos 1.29 version also work.

lordofhyphens, 2017-01-11:
99% certain it is not Slic3r but the library install on your system.

I haven't been able to reproduce on Debian at all, installing multiple
ways.

lordofhyphens, 2017-01-11:
I no longer see this on Windows (I believe root cause there was a broken
Perl).

lordofhyphens, 2017-01-11:
See #3540 as well for background. I think @bubnikv just did the work around
that forced an install of 0.67 in the build scripts in the prusa3d fork (if
I am wrong about this, please let me know).

When reporting what works and what doesn't,  you need to be very specific
(and sure) about the state of the system. I know that opengl 0.6704 works
with older versions of Slic3r's draw code. 0.70 had issues for which there
was a PR to fix those by rewriting the bed draw code to use VBOs. I did
report the issue upstream, never saw a reply.

You also should not trust the state of your Perl distro packages, older
versions of the build script, when run as sudo, can and will change things
behind your back. I don't know if the Prusa3d fork has incorporated the
local::lib change yet which avoids that issue.

alexxy, 2017-01-11:
I build slic3r using distro machinery. All packages are current. And all installed via package manager including slic3r (master version and prusa3d version).

lordofhyphens, 2017-01-11:
I can't support building with distro machinery. You need to contact the
maintainer of that macinery.

TheJJ, 2017-01-15:
After some investigation I found out: The `OpenGL.pm`, even in version `0.700.0`, creates this magic `AUTOLOAD` function that is [called each time](http://perldoc.perl.org/perlsub.html#Autoloading) a undefined function/variable is called/used.

The `OpenGL.pm` uses the `AutoLoader` in this handler function (the `sub AUTOLOAD`)
After tracing for what this function is then called, i found:

non-working version:
```
OpenGL::StructureNotifyMask: <<- 
OpenGL::GL_FLOAT: <<- 
OpenGL::GL_DEPTH_TEST: <<- 
OpenGL::GL_LEQUAL: <<- 
OpenGL::GL_CULL_FACE: <<- 
OpenGL::GL_BLEND: <<- 
OpenGL::GL_SRC_ALPHA: <<- 
OpenGL::GL_ONE_MINUS_SRC_ALPHA: <<- 
OpenGL::GL_LINE_SMOOTH: <<- 
OpenGL::GL_POLYGON_SMOOTH: <<- 
OpenGL::GL_MULTISAMPLE: <<- 
OpenGL::GL_LIGHT_MODEL_AMBIENT: <<- 
OpenGL::GL_LIGHTING: <<- 
OpenGL::GL_LIGHT0: <<- 
OpenGL::GL_LIGHT1: <<- 
OpenGL::GL_POSITION: <<- 
OpenGL::GL_SPECULAR: <<- 
OpenGL::GL_DIFFUSE: <<- 
OpenGL::GL_SMOOTH: <<- 
OpenGL::GL_FRONT_AND_BACK: <<- 
OpenGL::GL_AMBIENT_AND_DIFFUSE: <<- 
OpenGL::GL_SHININESS: <<- 
OpenGL::GL_EMISSION: <<- 
OpenGL::GL_COLOR_MATERIAL: <<- 
OpenGL::GL_LESS: <<- 
OpenGL::GL_COLOR_BUFFER_BIT: <<- 
OpenGL::GL_DEPTH_BUFFER_BIT: <<- 
OpenGL::GL_MODELVIEW: <<- 
OpenGL::GL_VERTEX_ARRAY: <<- 
OpenGL::GL_NORMAL_ARRAY: <<- 
OpenGL::GL_PROJECTION: <<- 
OpenGL::GL_QUADS: <<- 
OpenGL::glGenBuffersARB_p: 1<<- 
Trace begun at /usr/lib64/perl5/vendor_perl/5.24.1/x86_64-linux-thread-multi/OpenGL.pm line 6184
OpenGL::AUTOLOAD(1) called at /usr/lib64/perl5/vendor_perl/5.24.1/Slic3r/GUI/3DScene.pm line 859
Slic3r::GUI::3DScene::Base::Render('Slic3r::GUI::Plater::3D=HASH(0x375bdb0)', 'Wx::PaintDC=SCALAR(0x4878958)') called at /usr/lib64/perl5/vendor_perl/5.24.1/Slic3r/GUI/3DScene.pm line 117
Slic3r::GUI::3DScene::Base::__ANON__('Slic3r::GUI::Plater::3D=HASH(0x375bdb0)', 'Wx::PaintEvent=SCALAR(0x4878c88)') called at /usr/lib64/perl5/vendor_perl/5.24.1/Slic3r/GUI/MainFrame.pm line 75
eval {...} at /usr/lib64/perl5/vendor_perl/5.24.1/Slic3r/GUI/MainFrame.pm line 75
Slic3r::GUI::MainFrame::new('Slic3r::GUI::MainFrame', 'mode', 'expert', 'no_controller', 1, 'no_plater', undef) called at /usr/lib64/perl5/vendor_perl/5.24.1/Slic3r/GUI.pm line 140
Slic3r::GUI::OnInit('Slic3r::GUI=HASH(0x13e2318)') called at /usr/lib64/perl5/vendor_perl/5.24.1/x86_64-linux-thread-multi/Wx/App.pm line 36
eval {...} at /usr/lib64/perl5/vendor_perl/5.24.1/x86_64-linux-thread-multi/Wx/App.pm line 36
Wx::App::new('Slic3r::GUI') called at /usr/bin/slic3r.pl line 113
Goto undefined subroutine &AutoLoader::AUTOLOAD at /usr/lib64/perl5/vendor_perl/5.24.1/x86_64-linux-thread-multi/OpenGL.pm line 6188
```

working version:
```
OpenGL::StructureNotifyMask: <<- 
OpenGL::GL_FLOAT: <<- 
OpenGL::GL_DEPTH_TEST: <<- 
OpenGL::GL_LEQUAL: <<- 
OpenGL::GL_CULL_FACE: <<- 
OpenGL::GL_BLEND: <<- 
OpenGL::GL_SRC_ALPHA: <<- 
OpenGL::GL_ONE_MINUS_SRC_ALPHA: <<- 
OpenGL::GL_LINE_SMOOTH: <<- 
OpenGL::GL_POLYGON_SMOOTH: <<- 
OpenGL::GL_MULTISAMPLE: <<- 
OpenGL::GL_LIGHT_MODEL_AMBIENT: <<- 
OpenGL::GL_LIGHTING: <<- 
OpenGL::GL_LIGHT0: <<- 
OpenGL::GL_LIGHT1: <<- 
OpenGL::GL_POSITION: <<- 
OpenGL::GL_SPECULAR: <<- 
OpenGL::GL_DIFFUSE: <<- 
OpenGL::GL_SMOOTH: <<- 
OpenGL::GL_FRONT_AND_BACK: <<- 
OpenGL::GL_AMBIENT_AND_DIFFUSE: <<- 
OpenGL::GL_SHININESS: <<- 
OpenGL::GL_EMISSION: <<- 
OpenGL::GL_COLOR_MATERIAL: <<- 
OpenGL::GL_LESS: <<- 
OpenGL::GL_COLOR_BUFFER_BIT: <<- 
OpenGL::GL_DEPTH_BUFFER_BIT: <<- 
OpenGL::GL_MODELVIEW: <<- 
OpenGL::GL_VERTEX_ARRAY: <<- 
OpenGL::GL_NORMAL_ARRAY: <<- 
OpenGL::GL_PROJECTION: <<- 
OpenGL::GL_QUADS: <<- 
OpenGL::GL_ARRAY_BUFFER_ARB: <<- 
OpenGL::GL_STATIC_DRAW_ARB: <<- 
OpenGL::GL_TRIANGLES: <<- 
OpenGL::GL_LINES: <<- 
OpenGL::GL_ELEMENT_ARRAY_BUFFER_ARB: <<- 
OpenGL::GL_RGBA: <<- 
OpenGL::GL_UNSIGNED_BYTE: <<- 
```

-> the `glGenBuffersARB_p(1)` call does not need to be autoloaded in the working version. Why that is, I'm still figuring out.

But, `OpenGL.pm` is of course buggy to use the `AutoLoader` in its magic `AUTOLOAD` function without ever importing that perl module (`use AutoLoader`). So this looks like dead code that is triggered and fails only if other things are broken.

What causes this remains unclear, except that the `glGenBuffersARB_p` opengl function is somehow not defined at the point and therefore is autoloaded, which crashes.

lordofhyphens, 2017-02-22:
https://bintray.com/lordofhyphens/Slic3r/Slic3r_Dependencies/Dependencies-1.3.0-dev
has a copy of all of the dependencies used for a couple of Strawberry perl versions. 
Give one of those a try (depending on your used version of Strawberry perl). It's 64bit too now as soon as I verify that #3723 works and merge it in.

closed 2017-02-22 by lordofhyphens