slic3r/Slic3r #3601: The XS/C++ code failed to compile, aborting on MacOS Sierra

mlaustin6, 2016-12-03:
Slice3r version - 1.2.9-503-gfde6e2e
Hash value - fde6e2e61deabd097a3bf565d4948077147b25ab
MacOS Sierra 10.12.1

I'm having a problem compiling and getting the error show above.  I did the following on my mac.  I installed Xcode 8.1, which happens to come which happens to have command line tools installed.  I then installed homebrew and boost with this command:  install boost --universal.  I installed local::lib version 2.000019.  It was installed to my home folder.  I changed bash to the local-lib directory and rand these commands:

perl Makefile.PL --bootstrap
make test && make install
echo 'eval $(perl -I$HOME/perl5/lib/perl5 -Mlocal::lib)' >> ~/.bashrc
cd
source ~/.bashrc

Then I followed with a git pull and build with the following commands:

git clone git://github.com/alexrj/Slic3r
cd Slic3r
cpan App::cpanminus
perl Build.PL  

Here is what the build log says:

cpanm (App::cpanminus) 1.7042 on perl 5.018002 built for darwin-thread-multi-2level
Work directory is /Users/Mark/.cpanm/work/1480802775.14550
You have make /usr/bin/make
You have LWP 6.05
You have /usr/bin/tar: bsdtar 2.8.3 - libarchive 2.8.3
You have /usr/bin/unzip
--> Working on ./xs
Entering /Users/Mark/Slic3r/xs
Checking if you have Module::Build 0.38 ... Yes (0.4003)
Checking if you have ExtUtils::Install 1.46 ... Yes (1.59)
Configuring /Users/Mark/Slic3r/xs
Running Build.PL
-> OK
Checking dependencies from MYMETA.json ...
Checking if you have ExtUtils::Typemaps::Default 1.05 ... No
Checking if you have Test::More 0 ... Yes (1.302067)
Checking if you have ExtUtils::Typemaps::Basic 1.05 ... No
Checking if you have Module::Build 0.3601 ... Yes (0.4003)
Checking if you have ExtUtils::ParseXS 3.18 ... Yes (3.30)
Checking if you have ExtUtils::Typemaps 1.00 ... Yes (3.30)
Checking if you have ExtUtils::XSpp 0.18 ... Yes (0.18)
==> Found dependencies: ExtUtils::Typemaps::Default, ExtUtils::Typemaps::Basic
Searching ExtUtils::Typemaps::Default (1.05) on cpanmetadb ...
--> Working on ExtUtils::Typemaps::Default
Fetching http://www.cpan.org/authors/id/S/SM/SMUELLER/ExtUtils-Typemaps-Default-1.05.tar.gz
-> OK
Unpacking ExtUtils-Typemaps-Default-1.05.tar.gz
Entering ExtUtils-Typemaps-Default-1.05
Checking configure dependencies from META.json
Checking if you have Module::Build 0.40 ... Yes (0.4003)
Checking if you have ExtUtils::Install 1.46 ... Yes (1.59)
Configuring ExtUtils-Typemaps-Default-1.05
Running Build.PL
-> OK
Checking dependencies from MYMETA.json ...
Checking if you have Module::Build 0 ... Yes (0.4003)
Checking if you have ExtUtils::Typemaps 3.18_03 ... Yes (3.30)
Building and testing ExtUtils-Typemaps-Default-1.05
-> OK
Successfully installed ExtUtils-Typemaps-Default-1.05
Searching ExtUtils::Typemaps::Basic (1.05) on cpanmetadb ...
Already tried ExtUtils-Typemaps-Default-1.05. Skipping.
Building and testing Slic3r-XS-0.01
-> FAIL Installing ./xs failed. See /Users/Mark/.cpanm/work/1480802775.14550/build.log for details. Retry with --force to force install it.
1 distribution installed

How do I fix this so it compiles?

lordofhyphens, 2016-12-03:
You should look at the build log, there's nothing particularly useful here.

mlaustin6, 2016-12-03:
That is the build log I copied.  Do you mean I should copy the whole contents of what is in bash?

alexrj, 2016-12-04:
Interesting. This is the key part:

```
Can't load '/Users/Mark/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.bundle' for module Slic3r::XS: dlopen(/Users/Mark/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.bundle, 2): Library not loaded: libboost_system-mt.dylib
Referenced from: /Users/Mark/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.bundle
Reason: unsafe use of relative rpath libboost_system-mt.dylib in /Users/Mark/Slic3r/xs/blib/arch/auto/Slic3r/XS/XS.bundle with restricted binary at /System/Library/Perl/5.18/darwin-thread-multi-2level/DynaLoader.pm line 194.
```

alexrj, 2016-12-04:
It seems to be a pretty common issue on MacOS X >= 10.11:
http://stackoverflow.com/questions/31343299/mysql-improperly-configured-reason-unsafe-use-of-relative-path

@mlaustin6, try this: `SLIC3R_STATIC=1 perl Build.PL --xs`

mlaustin6, 2016-12-04:
That did not work.  I get the following log:

! Can't write to /Library/Perl/5.18 and /usr/local/bin: Installing modules to /Users/Mark/perl5
! To turn off this warning, you have to do one of the following:
!   - run me as a root or with --sudo option (to install to /Library/Perl/5.18 and /usr/local/bin)
!   - Configure local::lib in your existing shell to set PERL_MM_OPT etc.
!   - Install local::lib by running the following commands
!
!         cpanm --local-lib=~/perl5 local::lib && eval $(perl -I ~/perl5/lib/perl5/ -Mlocal::lib)
!
App::cpanminus is up to date. (1.7042)
Cleaning up build files
cannot unlink file for buildtmp/XS.o: Permission denied at /System/Library/Perl/5.18/Module/Build/Base.pm line 5289.
cannot restore permissions to 0100644 for buildtmp/XS.o: Permission denied at /System/Library/Perl/5.18/Module/Build/Base.pm line 5289.
Couldn't remove 'buildtmp/XS.o': Permission denied
cpanm (App::cpanminus) 1.7042 on perl 5.018002 built for darwin-thread-multi-2level
Work directory is /Users/Mark/.cpanm/work/1480820381.27736
You have make /usr/bin/make
You have LWP 6.05
You have /usr/bin/tar: bsdtar 2.8.3 - libarchive 2.8.3
You have /usr/bin/unzip
!
! Can't write to /Library/Perl/5.18 and /usr/local/bin: Installing modules to /Users/Mark/perl5
! To turn off this warning, you have to do one of the following:
!   - run me as a root or with --sudo option (to install to /Library/Perl/5.18 and /usr/local/bin)
!   - Configure local::lib in your existing shell to set PERL_MM_OPT etc.
!   - Install local::lib by running the following commands
!
!         cpanm --local-lib=~/perl5 local::lib && eval $(perl -I ~/perl5/lib/perl5/ -Mlocal::lib)
!
--> Working on ./xs
Entering /Users/Mark/Slic3r/xs
Checking if you have Module::Build 0.38 ... Yes (0.4003)
Checking if you have ExtUtils::Install 1.46 ... Yes (1.59)
Running Build.PL
Configuring /Users/Mark/Slic3r/xs ... Can't create '_build/runtime_params': Permission denied at /System/Library/Perl/5.18/Module/Build/Notes.pm line 110.
N/A
Checking dependencies from MYMETA.json ...
Checking if you have ExtUtils::ParseXS 3.18 ... Yes (3.30)
Checking if you have ExtUtils::Typemaps 1.00 ... Yes (3.30)
Checking if you have ExtUtils::Typemaps::Basic 1.05 ... Yes (1.05)
Checking if you have ExtUtils::XSpp 0.18 ... Yes (0.18)
Checking if you have Module::Build 0.3601 ... Yes (0.4003)
Checking if you have ExtUtils::Typemaps::Default 1.05 ... Yes (1.05)
Checking if you have Test::More 0 ... Yes (1.302067)
Building and testing Slic3r-XS-0.01 ... Can't create '_build/runtime_params': Permission denied at /System/Library/Perl/5.18/Module/Build/Notes.pm line 110.
FAIL
! Installing ./xs failed. See /Users/Mark/.cpanm/work/1480820381.27736/build.log for details. Retry with --force to force install it.
The XS/C++ code failed to compile, aborting

mlaustin6, 2016-12-04:
Not sure why the above lines are struck as I just copied and pasted.

lordofhyphens, 2016-12-04:
That's markdown for you. You can drag/drop zip files and I think txt files.

alexrj, 2016-12-04:
"Permission denied" suggests to me that you previously run as root, and you need to do the same again.

mlaustin6, 2016-12-04:
What happened was that I ran as non-root in the beginning and got the error.  Then I ran as root thinking since I had local-lib installed that maybe it needed root credentials.  It failed just like the first time.  So when adding your suggested changes, I guess it makes sense to run as root. I'm obviously not a programmer :). 

I really don't want to install as root.  Is there any danger to doing so? Can I reverse this to not install as root?  If that is my only option, I wouldn't need local-lib amd can delete it right?

lordofhyphens, 2016-12-04:
If you have local::lib installed and it's configured (which does need to be
installed as root IIRC), just ensure that your environment is set up
correctly to use local::lib, delete the Slic3r source folder and
re-checkout as you.

AFAIK the only "danger" to installing as root is that cpanm likes to stomp
over perl libs not installed by it; the only thing the installation process
needed root privs for is sticking the Slic3r XS library (which is the core
of the program) into the global Perl package dir. If you were using a
package system to maintain your Perl install (on OSX I think this is
unlikely but I don't really know). Slic3r will still be run as you.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

mlaustin6, 2016-12-04:
I don't know if it is installed as root.  I followed the instructions to download local::lib and unzipped it to my home folder.  I then changed the terminal director to users/mark/local-lib and ran these commands

perl Makefile.PL --bootstrap
make test && make install
echo 'eval $(perl -I$HOME/perl5/lib/perl5 -Mlocal::lib)' >> ~/.bashrc
cd
source ~/.bashrc

There is no sudo command there, so I doubt it is installed as root.  Should I run sudo perl Makefile.PL --bootstrap instead?

lordofhyphens, 2016-12-04:
You're probably fine. Delete the Slic3r dir and start from a new checkout.

mlaustin6, 2016-12-04:
So I deleted the Slic3r director and tried again.  I ran SLIC3R_STATIC=1 perl Build.PL --xs, and I have new errors.  It doesn't seem like the xs error is there.  Here is the log.
[slic3r build error.pdf](https://github.com/alexrj/Slic3r/files/629790/slic3r.build.error.pdf)

lordofhyphens, 2016-12-04:
Looks like something else went weird with the tests, but the important part
did compile and install.

mlaustin6, 2016-12-04:
ok, so i will build the gui part and see how it works.  To run the gui, do I need issue a perl slic3r.pl command in terminal?

mlaustin6, 2016-12-04:
Next problem.  I had an error with the WX module, so I ran this command per the wiki, brew install wxmac.  Here is that log.
[slic3r gui build error.pdf](https://github.com/alexrj/Slic3r/files/629819/slic3r.gui.build.error.pdf)

lordofhyphens, 2016-12-04:
GL building wxperl, it's the bane of many people's existence. Best I can
say about that. -.-;

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

mlaustin6, 2016-12-04:
So I found out something interesting.  I created a sierra vm so I could test this out as root.  I did all on the wiki and ran sudo cpan App::cpanminus.  It asks you if you want to do install as the following: local::lib, sudo, or manual.  I obviously picked sudo for this install.  I got the xs error on the first one, so I tried sudo SLIC3R_STATIC=1 perl Build.PL –xs.  All tests were successful.  Running the gui command as sudo, wx failed to install.  So I did a brew install wxmac.   I ran the gui command as sudo again,  and wx failed to install.  But this time, running perl slic3r.pl give me slic3r options but not from a gui perspective.  

On my non vm machine, should I run the sudo sudo cpan App::cpanminus and choose the local::lib option after removing the existing local-lib folder?  Then I can exit terminal and come back in to run non-sudo installs.  

Regarding the wx failing to install, how do I get it to work?  Or am I just hosed on a mac?

bubnikv, 2016-12-04:
Don't know whether you tried that, but we run a OSX builds of Slic3r using
a perlbrew - a user installed perl. This makes the slic3r build process not
interfere with the system wide perl.

mlaustin6, 2016-12-04:
I have not tried perlbrew.  I'm not sure how to install that.  The may fix the build process but will most likely continue to give the wx failure.

mlaustin6, 2016-12-05:
We can close this issue out.  I gave up on installing it in Mac.  In the past, I could not get it to install on Ubuntu 16.10.  This time it did, so I will use that.  The only problem I am having is I cannot get the shortcut to work and need to start slic3r from terminal.  I used this shortcut in a text file:

#!/bin/sh
eval $(perl -Mlocal::lib)
perl ./slic3r.pl $*

I then modded it with chmod +x slic3r. I get invalid url given, exiting, no '.' in the uri.  Any suggestions on fixing this?  Or do I need to open a new issue?

lordofhyphens, 2016-12-05:
I'd use the full path to slic3r.pl in the script, because that's relative
to your CWD.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

mlaustin6, 2016-12-05:
i still get the error using /home/.../slic3r.pl.  It's not a big deal I guess.  I'll just do it through terminal.

lordofhyphens, 2017-03-05:
abe22fa4068998bef87a5525bb588c3fcf1a0d96 should fix this. 

Closing for now, please try and see ;) 

Also, the build server is partially up and running now (at least, I have made and packaged builds on it). 

https://bintray.com/lordofhyphens/Slic3r/download_file?file_path=slic3r-1.3.0-36e1489-osx-package.dmg

lordofhyphens, 2017-03-05:
Please make comments about autobuild in #3738, only post here if still seeing that oddball build issue that was reported up top.

closed 2017-03-05 by lordofhyphens