slic3r/Slic3r #3603: CPU usage bump when hovering the 3D preview aera

gege2b, 2016-12-05:
Hi all

### Version
1.2.9 x64 stable for windows

### Operating system type + version
Windows 10 x64 

### Behavior
Pretty hight CPU load when hovering the mouse over the preview window  (at least, higher than expected)

Load a STL and wait for the background slicing to process

Open the task manager (or other ressource viewing software) and continuously move the mouse over the preview windows, you'll see a jump in CPU% (above 30% in my case) without visible reason (no redraw/refresh of the screen, etc...)

If you stop moving the mouse the CPU% will drop to < 1%
If you move the mouse over the right panel, CPU% will drop too
the problem is less visible on 2D and Layer views (CPU% jump to ~3-4%) or 3D view (~15% )

It is model and config independent

This is not a very visible problem, as it will not slow down slic3r or the computer too much, but more CPU usage means more eat and power consumption, could be problematic on netbook

biocomp, 2016-12-05:
I guess I'm experiencing the same (or similar problem). With fresh slic3r build (commit fde6e2e61deabd097a3bf565d4948077147b25ab), I see flickering whenever mouse moves over preview area. The flickering is very visible, it's like the screen is filled with white first, then actual image is drawn on top of that.

lordofhyphens, 2016-12-05:
This could be related to how I implemented the fix for OpenGL.pm crashing.
I instantiate some buffer objects for the background lines and points but
then have to put everything back when I'm done because I was avoiding
rewriting all of the draw code.

Dig through and find 3Dscene.pm and change the value of HAS_VBO to 0 and
see if the problem still shows up (assuming it doesn't crash of course :) )
That will revert back to the old behavior.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2016-12-06:
> I changed the value to 0, flickering was still there and no crashes.

Dang. I was hoping that was it :(

gege2b, 2016-12-06:
As mentioned in my report, I use the 1.2.9 stable, witch was released one and a half year ago
Anyway, I see no flickering at all when moving the mouse over the preview window

When I got some time, I'll try to compile slic3r under windows, but the last time I tried to it failed and I gave up

gege2b, 2016-12-06:
I got the same behaviour on linux (debian jessie) with current head
BUT, it doesn't go above ~10% (I'm using KSysGuard to look at this)

Another thing I spotted is that the % depends on the quantity of lines present in the  preview

Well, it's hard to test because of #3589 , as soon as I choose a higher infill density, *BAM*  => division by zero

lordofhyphens, 2016-12-06:
If you have the preview window open immediately when slicing it will crash
due to assumptions made for viewport code.

alexrj, 2016-12-21:
Okay, now there shouldn't be any more CPU usage bump when hovering the 3D preview (but still when hovering the 3D plater).

gege2b, 2016-12-23:
seems to work on my debian, still need to test it on my win10's office computer (where the effect was clearly noticeable), after the new year :)

lordofhyphens, 2017-02-07:
@gege2b Any update? Can I close this?

gege2b, 2017-02-07:
well, sorry I forgot to check

it's seems ok now, the bump is still here but around 3-5% instead of 30%+ so I guess you can close this

and thank you !