slic3r/Slic3r #3604: Second Extruder will only print supports

TimeSentinel, 2016-12-05:
### Version
1.2.9 - also tried 1.1.7

### Operating system type + version
Windows 10 home 64-bit

### Behavior
* Set capabilities to 2 extruders, however when I set print options to print all perimeters, infill and supports on extruder 2, it slices for extruder 1 for main part and will only slice supports to extruder 2. If I turn off supports, it will use ext 2 just fine (unfortunately, my ext 1 is not currently working and I need to print with supports). 
If I set ext 1 to print supports and ext 2 to print everything else, it still changes to ext 2 printing supports and ext 1 printing everything else.
* I have tried reinstalling both Repetier and Slic3r and different versions. Tried creating new settings for printer, filament and print. I even tried installing on another computer and slicing - same issue. Tried slicing with slic3r outside of repetier; same result. Why can't I print the entire file with supports to ext 2?

Am I missing something? Is there a workaround (I have found that if I print the whole file to ext 1, then add a T1 line at the beginning of the file it seems to work - but should I have to?)?

No files or screenshots at this time - not in front of my laptop.

lordofhyphens, 2016-12-05:
Can't do much until you provide at least an exported config and model.

Something you could check is the Part Settings for a model (where you can
set the extruder for individual parts).

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

TimeSentinel, 2016-12-06:
[image]
[image]
[image]
[image]
[image]

TimeSentinel, 2016-12-06:
[config-gcode.zip](https://github.com/alexrj/Slic3r/files/632781/config-gcode.zip)

TimeSentinel, 2016-12-06:
Let me know if I can provide anything else. Right now I have to slice everything to Ext 1, then add T1 to the beginning of the file (or close to the beginning). 
The attached files are just one of the methods I have experimented with. I have tried variations of ext1 and 2 in various positions - it either comes out with everything under ext 1 or only supports under ext 2.

lordofhyphens, 2016-12-07:
FYI, it needs to be `File->Export Config`, not `File->Export Config Bundle`. The former makes a profile that is exactly what you were using, the latter is everything available to you.

lordofhyphens, 2016-12-07:
Still need a test model.

lordofhyphens, 2017-01-26:
I also need you to use the Slic3r UI directly, not RepetierHost. We can't support use through RepetierHost, so you need to replicate the problem using Slic3r's own GUI.

Closing this for now due to non-reproducibility. If  you provide what was asked, I will reopen.

I also suggest using the development build (linked through https://github.com/alexrj/Slic3r/blob/master/README.md), it's had a lot of fixes since 1.2.9. Extract and run slic3r.bat.

closed 2017-01-26 by lordofhyphens