slic3r/Slic3r #3605: Can't locate object method "new" via package "LWP::UserAgent" (perhaps you forgot to load "LWP::UserAgent"?) at /usr/share/perl5/vendor_perl/Slic3r/GUI/Plater.pm line 1330.

ioquatix, 2016-12-07:
### Version
1.3.0-dev

### Operating system type + version
Arch Linux

### Behavior
- Any model, send to printer.

slic3r.pl crashes with the following output:
Can't locate object method "new" via package "LWP::UserAgent" (perhaps you forgot to load "LWP::UserAgent"?) at /usr/share/perl5/vendor_perl/Slic3r/GUI/Plater.pm line 1330.

lordofhyphens, 2016-12-07:
Is this a recent checkout?

ioquatix, 2016-12-07:
yes like a few days ago

lordofhyphens, 2016-12-07:
Yeah, apparently that got added w/o adding it as a ref and it works on some systems and not others (go figure). 
I'm testing a fix real quick (just add 'use LWP::UserAgent' to the top of the file with the other use statements to that file).

lordofhyphens, 2016-12-07:
Should be fixed now.

PR #3608 (2016-12-07): Add LWP::UserAgent to use of plater.pm
Fixes #3605
closed 2016-12-07 by lordofhyphens