slic3r/Slic3r #3606: UI hang when scaling or deleting 1 of many parts?

ioquatix, 2016-12-07:
### Version
1.3.0-dev from 4 days ago

### Operating system type + version
Arch linux

### Behavior
Download this model.
http://www.thingiverse.com/thing:1557428

Add Body.stl, Wheels.stl and Rims.stl

Try to resize Body part. UI will hang.

ioquatix, 2016-12-07:
Just adding Body.stl and scaling to 30% gives the following

```
Illegal division by zero at /usr/share/perl5/vendor_perl/Slic3r/GUI/3DScene.pm line 379.
Perl exited with active threads:
	1 running and unjoined
	0 finished and unjoined
	0 running and detached
```

lordofhyphens, 2016-12-07:
Turn off background processing. Also, see #3589 and #3591 (merge the changes from the PR to get a workaround for the divide-by-zero).

ioquatix, 2016-12-07:
Awesome. So should we close this issue?

lordofhyphens, 2016-12-07:
Also probably related to #3499. It can be closed, just as soon as I find something else to reference (as there's two issues here, both of which are duplicates of something else).

lordofhyphens, 2016-12-07:
Going to close, the former is related to Perl threads (which will go away once we replace the GUI) interacting badly with libslic3r, latter has a workaround (and has been reported elsewhere).

closed 2016-12-07 by lordofhyphens

ioquatix, 2016-12-09:
Great, thanks :)