slic3r/Slic3r #3607: "FLIP" is the wrong word.

cameronbergh, 2016-12-07:
### Version
129

mac, linux, windows

This is some excellent programming, 
Seriously, world changing stuff that you are doing. great job.

My only issue is a very small one.
When i right click on an object there is this option to "flip"
an object along a specified axis, however, this does not flip the object.
instead it reflects the object.

# let me demonstrate:

[image]

### above: the feature i am referring to
### below: an unmodified object which i am hastily trying to get printed.

[image]

### below: the object after being "flipped" upon the z axis (you mean reflected)
### notice that the threads now turn in the opposite direction, making this printed part useless.

[image]

###here is a picture of a girl doing a flip:

[image]
http://www.wikihow.com/Do-a-Triple-Front-Flip

### here is what the "flip" function in slic3r is actually doing:
this image is from the wikipedia page for reflection!
https://en.wikipedia.org/wiki/Reflection_(mathematics)
[image]

one could argue that if i was a smart man, I would have used the rotate function to accomplish what I had intended, but i was in a hurry and didn't notice the subtle difference.

slic3r is still the best software for this task
thanks for all your hard work!

-Cameron

https://sciencelens.files.wordpress.com/2012/06/coin_flip_s.jpg

lordofhyphens, 2016-12-07:
This was already fixed in -dev a year and a half ago. #3060

closed 2016-12-07 by lordofhyphens

cameronbergh, 2016-12-07:
LOL my bad thanks

lordofhyphens, 2016-12-07:
And you'd put a lot of work into it too.