slic3r/Slic3r #3611: Move ConfigOption to XS/C++

lordofhyphens, 2016-12-12:
Leaving this here for @bubnikv and @alexrj as a FYI that I am working on porting the ConfigOption and Field packages to C++.

Needs:
* Implementation of the appropriate design pattern (thought process is currently [Builder](https://en.wikipedia.org/wiki/Builder_pattern) )
* XSP interface to the Perl backend (unnecessary after everything moves to C++)
* Support same calling convention as current Perl class.

Notes:
* Template classes + boost::any looks like it will handle polymorphism-related problems

lordofhyphens, 2016-12-12:
I will probably not maintain the same function names for everything, instead treat it more like a STL container and use the XSP interface to avoid changes to calling code.

bubnikv, 2017-12-01:
@lordofhyphens FYI, we have a new team member at Prusa Research - @YuSanka working on porting the Print / Filament / Printer tabs into C++.

lordofhyphens, 2017-12-01:
I will get what I have done committed this weekend then.

I should have a working port, I have not had the time to fully implement
all of the types and the XS interface and work out that plumbing.

bubnikv, 2017-12-01:
@lordofhyphens @YuSanka Thanks for the heads up.

YuSanka, 2017-12-05:
@lordofhyphens, I'm new team member in Prusa Research. I'll working on porting GUI into C++. For this moment I have a something in my repository https://github.com/prusa3d/Slic3r/tree/gui_translate_to_cpp. Look at this when you'll have a time. And commit, please, your changes if you have it.

lordofhyphens, 2017-12-07:
@YuSanka  https://lordofhyphens@github.com/lordofhyphens/gui-mockup is where my existing work lives. I've been swamped with my own day job and haven't been able to get back to it.

lordofhyphens, 2017-12-07:
@YuSanka I did get it to bind to libslic3r; the general architecture upstream is to keep libslic3r and the gui separate and as much gui-specific dependencies out of libslic3r. 

Frankly on my far-future to-do list is using something like SWIG templates to permit people to integrate libslic3r into their language of choice.

YuSanka, 2017-12-07:
@lordofhyphens  @bubnikv  Thanks a lot.