slic3r/Slic3r #3612: Gap fill creating unreasonably thin extrusions

VanessaE, 2016-12-12:
### Version

11585b88d5a1063f28a17ae5a90698e2c23ddcda

### Operating system type + version

Debian 8.6

### Behavior

When slicing some objects, gap fill is created that's way too thin for the configured nozzle diameter.

Since you can't reliably print a line thinner than the nozzle, and gap-fill is usually accompanied by retracts/restarts, you get a slightly-blobby, over-extruded line instead.  I guess the underlying layer "drags" extra plastic out of the nozzle, so if the next lines are supposed to be thick, they end up under-extruded slightly.

The attached model below only allows one perimeter between the interior of the model and the outside, so there's nowhere else for the extra plastic to go.

I think what's needed here is a configurable lower limit to gap fill width.  Anything smaller should either be omitted entirely, or perhaps those spaces should be filled by altering the width and position of the offending perimeter segments (even if it means overriding a user-configured static value).

Example screenshot below. Look between the middle circle and its two neighbors, and between the left-most circle and the straight line near it.  The thinnest of these segments is about 0.05 mm wide (by measuring from a screenshot of a much more zoomed-in view).  If you look below the middle circle, you can see a more reasonable gap fill width.
[image]

#### STL/Config (.ZIP) where problem occurs
http://www.thingiverse.com/thing:1363023
[Slic3r_config_bundle_1.3.0-dev-2016-12-11.ini.zip](https://github.com/alexrj/Slic3r/files/644932/Slic3r_config_bundle_1.3.0-dev-2016-12-11.ini.zip)
Use "Single material, 60 mms, 0.2mm layers" print profile.  I guess the filament profile and printer profile don't matter here (I was using "Inland Yellow", and there's only one printer profile).

lordofhyphens, 2016-12-12:
I thought part of the point of gap fill is that it's constrained between two perimeters and so you can usually get thinner areas filled in than you would normally be able.

VanessaE, 2016-12-12:
That may be, but I'm at a loss to explain the over-extrusion happening in those areas.  Perhaps those super-thin lines run up against a firmware limitation?

lordofhyphens, 2017-01-25:
Could be, it may be instructive to identify the gcode produced for those bits and see how the firmware interacts with that.

lordofhyphens, 2017-01-27:
I think the correct solution at this point is to identify the perimeters adjoining areas needing  gap fill and to shift them out and expand slightly  to cover the gaps instead as a pre-pass (assuming you don't go over some maximum width), then apply gap fill.

VanessaE, 2017-01-27:
That's what I was thinking, too.

bubnikv, 2017-01-27:
The same applies for the infills. Also the infills should be calculated before the gap fill. Currently it works vice versa,

1) perimeters are calculated
2) infill areas are calculated from the perimeters, rounding the infill areas
3) gaps are filled between the perimeters and the infill areas
4) infills are filled lines with lines of constant width, creating new gaps.

it should be vice versa:

1) perimeters are calculated
2) infill should be calculated from the perimeters without rounding the areas
3) infills should be filled with lines of variable width, creating as little gaps as possible
4) gaps should be filled between the perimeters and the infill lines.

The second solution is naturally significantly more difficult to implement and will run significantly slower.

VanessaE, 2017-05-20:
This doesn't seem to be much of a problem anymore.

closed 2017-05-21 by VanessaE