slic3r/Slic3r #3613: Slic3r hangs with "Processing triangulated mesh..."

kurthuwig, 2016-12-12:
### Version
1.2.9

### Operating system type + version
Kubuntu 16.04.1 LTS

### Behavior
* add 'g-doesntwork.stl' to the plater
* expected: slic3er slices the STL
* observed: slic3er stays in "Processing triangualted mesh..."

This happens with 'g-doesntwork.stl', but not with 'g.stl'. The difference of the two files is the depth of the emboss of the "G": in 'g.stl' it is larger than on 'g-doesntwork.stl'.

As a side mark: the first layer above the G uses the longer axis. Is there a way to turn it 90°, so that it has shorter bridges?

kurthuwig, 2016-12-12:
[slicing-hangs.zip](https://github.com/alexrj/Slic3r/files/645285/slicing-hangs.zip)

lordofhyphens, 2016-12-13:
Whatever it is, it's related to your fixed widths for everything. Once I unset all of that (set first layer to 200%) it sliced. 
This config does work (for my own printer): [config-working.zip](https://github.com/alexrj/Slic3r/files/647787/config-working.zip)

lordofhyphens, 2016-12-13:
It's the perimeters and default extrusion width both being 0.5 together. Slic3r wants to set the regular perimeter width to 0.85mm (which is too wide to form loops on the upper rim, so gap fill is used to handle that).

This is probably some kind of edge case that is being run into during perimeter generation (1.3.0-dev). 

The inflection point seems to be around a perimeters setting of 0.6005 on the provided model. 
I've found this with some other models with auto-calculation (can't post because licensing).

lordofhyphens, 2016-12-13:
It's the width + "extra perimeters where needed" option.

lordofhyphens, 2016-12-13:
From an investigation of make_perimeters(), perimeters added to the slice at hand aren't considered in the condition whether or not to make more perimeters. I'll have a fix up in a PR in a few minutes once I get it cleaned up properly.

lordofhyphens, 2016-12-13:
@kurthuwig With 1.2.9, disable extra perimeter detection as a workaround.

alexrj, 2016-12-16:
Fixed! Thank you @kurthuwig and @lordofhyphens.

closed 2016-12-16 by alexrj

bubnikv, 2017-02-02:
@lordofhyphens great catch!