slic3r/Slic3r #3615: Missing icons with 1.30_dev W7/64

justmyopinion, 2016-12-14:
### Version
  1.30_dev
### Operating system type + version
W7/64
### Behavior
When compiled (Citrus Perl 5.14 I do get a success but get a flickering display at plater with mouseover and some icons missing. Slic3r slices but crashes very easily with config changes.
[image]

lordofhyphens, 2016-12-14:
There's multiple issues here.

The icons being missing is probably a result of Slic3r not being able to
find its var directory. It uses the FindBin package to do this to get
relative to the script. If you moved Slic3r.pl that would break things.

Odds are the crashes are the known viewport issue  (turn off background
slicing and avoid looking at the preview at the beginning of a slicing
process... or take a look at #3591 )

The flickering sounds related to #3603. No idea why at this point, I
haven't been able to reproduce on my systems. Could also be related to your
graphics card and how I implemented a fix. If you go into 3Dscene.pm and
change HAS_VBO =0 you will get the old behavior which may work for you

lordofhyphens, 2016-12-14:
Caveat: I don't use CitrusPerl and never will.

justmyopinion, 2016-12-14:
Ok thanks

I run Slic3r from the compiled dir  \dev\slic3r, could it get confused during compile time?
I usually turn off background when crash problem is severe.

rel flickering:
Repetier Host reports this: ((reduces form)
16:26:16.669 : OpenGL version:3.3.0
16:26:16.669 : OpenGL renderer:GeForce 9800 GT/PCIe/SSE2
16:26:16.669 : Using fast VBOs for rendering is possible

So I would think VBO would work as it does very well in RH

lordofhyphens, 2016-12-14:
That path isn't figure out at compile time, it is run time.

lordofhyphens, 2017-01-25:
The missing icons are definitely a path issue in that Slic3r can't find its ./var directory. 

Can't tell you what crashes are (other than extra weirdness brought about by background processing, which I thoroughly dislike as an option).

We are using VBOs now for the background grid, doing everything with VBOs is on my to-do list.

lordofhyphens, 2017-01-26:
Since you're on Windows, you can give the Bintray build a whirl (see https://github.com/alexrj/Slic3r/blob/master/README.md for links).

There's another workaround for the icons which is to edit slic3r.pl and change the line with "var" near the beginning and replace "var" with the full path to where the slic3r/var directory is on your system. 

I'm working again on getting the x64 builds going again. 

Going to close this as I haven't been able to reproduce on Strawberry Perl and there is a workaround.

closed 2017-01-26 by lordofhyphens

justmyopinion, 2017-01-26:
I did try with Strawberry but had same result.
Searched for /var dir in slic3r.pl but did not find ref.
I will stick with curr or bintray.

Thankyou for your effort

lordofhyphens, 2017-01-26:
Just fyi: it is https://github.com/alexrj/Slic3r/tree/master/var that it is
looking for (relative to Slic3r.pl)