slic3r/Slic3r #3616: weird gcode outside of model

kameamea, 2016-12-14:
### Version 
1.2.9
### Operating system type + version
ubuntu 16.04

### Behavior
the top three levels of the gcode are outside of the stl model.
See the attached screenshots. I checked with blender that the stl is a manifold.
I expect the outside of the gears to be in a straight line.

[image]
[image]

#### STL/Config (.ZIP) where problem occurs

unfortunately github says "We don’t support that file type" when I try to upload the zip file. Hopefully I can attach it later on.

lordofhyphens, 2016-12-14:
Github may be acting case-sensitive. ".zip" should work as an extension (I
don't know if it extracts it or performs any other verifications).

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

kameamea, 2016-12-14:
I tried .zip and .ZIP  and different content with no success. May I email the zip file to you so you can attach it?

lordofhyphens, 2016-12-14:
Yeah, that's fine. Use the email on my profile.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2016-12-15:
Not seeing it in 1.3.0-dev, so that's something. 

Turn off thin wall detection as a workaround.

[image]

[image]

lordofhyphens, 2017-01-25:
No other comments from @kameamea and I couldn't reproduce and it's been over a month. Closing.

closed 2017-01-25 by lordofhyphens