slic3r/Slic3r #3617: Division by zero

gege2b, 2016-12-14:
### Version
current dev (as of 3a3b24ec7c803c58919a5401e5804261999875b4 )

### Operating system type + version
Debian 8.6 x64

### Behavior
It's an issue pretty similar to #3589 except one thing : I don't have to switch to 3D preview for Slic3r to hangs.

All I do is load the STL (witch have some errors but I don't think it's the problem here), let the background slicing to complete and simply change infill % from default 20 to 25% (or 30 or 15 or whatever). It hangs instantly.
weirdly, if I simply change the type of infill it doesn't hangs

So, right now, I just can't change the infill % 

Anyway If I change the infill density and then load a model, it's all right.

```
(note : the four folowing lines may vary or simply vanish. The last is always here.)
  Unable to close this loop having 26 points
  Unable to close this loop having 167 points
  Unable to close this loop having 1 points
  Unable to close this loop having 4 points
Illegal division by zero at /home/gege/3Dprint/Slic3r/lib/Slic3r/GUI/3DScene.pm line 379.
```

#### STL/Config (.ZIP) where problem occurs
STL : http://www.thingiverse.com/thing:1727943 (my daugher loves Skye ^_^ ) in this case. But I also tried  with other error-free STLs with same results
Default config

####Note : Maybe this is the same issue than the one mentioned above, but it appears when doing something else, that's why I opened a new issue. Don't hesitate to close it if it was unnecessary

lordofhyphens, 2016-12-14:
Make absolutely sure that  your model is manifold. "Unable to close loop"
errors are usually as sign that something else is wrong.

The illegal division by zero, however, is definitely related to the
bounding box issue.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

gege2b, 2016-12-15:
Yeah, that's why I tested other STL witch are manifold (no error and already printed in the past)

OK, so except for the non manifold and the "loop" errors witch are unrelated, I still get the division by zero error.

I'll close this issue as it's a duplicate of #3589 (possibly fixed by  #3591 I guess) , feel free to reopen it if needed for any reason

closed 2016-12-15 by gege2b

gege2b, 2016-12-18:
Allright, problem gone since #3591 has been merged :)