slic3r/Slic3r #3622: The XS/C++ code failed to compile, aborting

andihas, 2016-12-18:
### Version
1.3.0-dev
b5940e542a852f43e9991c711f02ac062cb2f9fc

### Operating system type + version
Linux Mint 18 Sarah, 4.4.0-53-generic x86_64

### Behavior
 cannot compile after commit e5412b5d69dc4d5415c92a7daaa47fb4860ca4ea

sudo perl Build.PL 
```
Don't worry, this module is optional.
Don't worry, this module is optional.
Don't worry, this module is optional.
cpanm (App::cpanminus) 1.7042 on perl 5.022001 built for x86_64-linux-gnu-thread-multi
Work directory is /home/andi/.cpanm/work/1482059129.6357
You have make /usr/bin/make
You have LWP 6.15
You have /bin/tar: tar (GNU tar) 1.28
Copyright © 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL Version 3 oder später <http://gnu.org/licenses/gpl.html>
Dies ist freie Software: Sie dürfen sie ändern und weiter verbreiten.
Es gibt KEINERLEI GARANTIE, so weit das Gesetz es erlaubt.

Geschrieben von John Gilmore und Jay Fenlason.
You have /usr/bin/unzip
--> Working on ./xs
Entering /home/andi/3d_printer/slic3r_mod/Slic3r/xs
Checking if you have Module::Build 0.38 ... Yes (0.4216)
Checking if you have ExtUtils::Install 1.46 ... Yes (2.04)
Running Build.PL
Configuring /home/andi/3d_printer/slic3r_mod/Slic3r/xs ... N/A
Checking dependencies from MYMETA.json ...
The XS/C++ code failed to compile, aborting`
```

lordofhyphens, 2016-12-18:
That log is unfortunately not very helpful. You can run `perl Build install` in the XS dir to build the important stuff.

Stop using sudo, that commit removes the need to do so. I just hope cpanm hasn't done too much damage to your local perl.

You probably should delete the entire dir and re-checkout as a regular user (not sudo). Then try a complete rebuild in that directory. This should avoid any weird permission issues coming from a dirty dir.

Do install local::lib either via cpanm or your package manager (see wiki for cpanm).

andihas, 2016-12-18:
Deleting the Folder {$HOME}/perl5 and building (without sudo) form a fresh checkout solved the compiling error.
Thanks

closed 2016-12-18 by andihas