slic3r/Slic3r #3623: TEST dependency fails with the latest pull update

Tinchus2009, 2016-12-18:
### Version
Latest pull

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
win10 64

### Behavior
I just did a git pull, it seems to compile ok, except for a dependency, TEST-SIMPLE wich fails to install. Still, the build finishes and then issuing a perl buil.pl --gui finishes the job.
IS this test-simple important. 

Ataches logs

How can I install (force install) the test-simple throug cpan manually? 
[build.zip](https://github.com/alexrj/Slic3r/files/659440/build.zip)

lordofhyphens, 2016-12-18:
Probably not important :)

Tinchus2009, 2016-12-18:
I "somehow" solved it: I just issued a cpan testsimple command and the test module installed with no problems this time, after that the build of slicer completed perfectly

closed 2017-01-25 by lordofhyphens