slic3r/Slic3r #3624: All test failling

gege2b, 2016-12-18:
Hi there, I just ran into a nice build issue right now :)
seems to be pretty similar to #3622 but I gave more info on what's going on

I saw later that sudo is not needed anymore (and of course, I used the --sudo), in the missing stuff at the begining of the log, cpanm was upgraded

I tried to delete my entire local repo and re-pull it, same problem

### Version
_Version of Slic3r used goes here_
807d441c6ea5b0b149649bae4dc4fe8a03239c16

### Operating system type + version
Debian 8.6 x64

### Behavior
The perl stuff is building/testing fine, then comme the XS stuff, and albait the build seem to go fine, all the test are failling

```

(... all went fine here ...)

t/23_config.t ................... ok   
All tests successful.
Files=22, Tests=495,  1 wallclock secs ( 0.09 usr  0.02 sys +  1.01 cusr  0.06 csys =  1.18 CPU)
Result: PASS
Building Slic3r-XS
Files found in blib/arch: installing files in blib/lib into architecture dependent library tree
Installing /home/gege/3Dprint/Slic3r/local-lib/lib/perl5/x86_64-linux-gnu-thread-multi/auto/Slic3r/XS/XS.bs
Installing /home/gege/3Dprint/Slic3r/local-lib/lib/perl5/x86_64-linux-gnu-thread-multi/auto/Slic3r/XS/XS.so
Installing /home/gege/3Dprint/Slic3r/local-lib/lib/perl5/x86_64-linux-gnu-thread-multi/Slic3r/XS.pm
OK
Successfully installed Slic3r-XS-0.01
1 distribution installed
t/angles.t ..................... Class::XSAccessor exists but failed to load with error: Perl API version v5.14.0 of Class::XSAccessor does not match v5.20.0 at /usr/share/perl/5.20/XSLoader.pm line 117.
Compilation failed in require at /home/gege/perl5/lib/perl5/Module/Runtime.pm line 317.
Odd number of elements in hash assignment at /usr/local/share/perl/5.20.2/Moo.pm line 172.
Use of uninitialized value in list assignment at /usr/local/share/perl/5.20.2/Moo.pm line 172.
Use of uninitialized value in subroutine entry at /usr/local/share/perl/5.20.2/Method/Generate/Constructor.pm line 78.
Use of uninitialized value $package in concatenation (.) or string at /usr/local/share/perl/5.20.2/Method/Generate/Constructor.pm line 101.
Can't call method "register_attribute_specs" on an undefined value at /usr/local/share/perl/5.20.2/Method/Generate/Constructor.pm line 250.
Compilation failed in require at /usr/local/share/perl/5.20.2/Moo.pm line 170.
Compilation failed in require at /home/gege/perl5/lib/perl5/Module/Runtime.pm line 317.
Compilation failed in require at /home/gege/3Dprint/Slic3r/t/../lib/Slic3r.pm line 58.
BEGIN failed--compilation aborted at /home/gege/3Dprint/Slic3r/t/../lib/Slic3r.pm line 58.
Compilation failed in require at t/angles.t line 15.
BEGIN failed--compilation aborted at t/angles.t line 15.
*** Error in `/usr/bin/perl': free(): invalid pointer: 0x00000000024bc00c ***
t/angles.t ..................... No subtests run 

(.... a VERY large amount of similar lines ...)

```

Here is the complete log (almost, some stuff are missing from the begining because of buffer size on my terminal window)
https://framabin.org/?0290aa68ef7038c6#LSxPRgE13nMWuIPs5Z/7VigJwIQ/RVZ9P7iaI6er5uM=

According to the log, there's a mismach between the XSAccessor and the perl version isntalled on my system

I've never encoutered this problem until now

#### STL/Config (.ZIP) where problem occurs

not relevant

lordofhyphens, 2016-12-18:
> RVZ9P7iaI6er5uM=

Delete your build dir and do a fresh checkout.

gege2b, 2016-12-18:
Already tested as mentionned and it didn't work

BUT I just tested by moving away my {$HOME}/perl5 dir and it worked fine

If this is related to the local::lib change  ( #3619 ) and the sudo being not needed anymore, I think something should be done to avoid things like this to happen (like ignore the --sudo parameter and/or show a warning ?)

I keep this issue open, feel free to close it if needed :)

lordofhyphens, 2016-12-18:
@alexrj any thoughts

lordofhyphens, 2016-12-18:
95% certain it is local lib commit. Remove your stuff added to .bashrc,
that is likely the culprit if $home/perl5 moving sorted it out.

alexrj, 2016-12-18:
Hm, interesting, I wonder where does that Perl 5.14 come from, since it looks like your system Perl is 5.20.

Do you have multiple perls installed? With perlbrew, maybe?

gege2b, 2016-12-18:
perl 5.20 is the only version of perl I have (and I never installed/removed another flavour of perl in the past neither)

lordofhyphens, 2017-01-25:
Yeah, I can't reproduce this either. Closing. 

I suggest doing a full purge of your Slic3r source directory and checking out a new version to ensure that there isn't anything weird going on.' Also remove any eval(perl -Mlocal::lib) commands from a .bashrc and logoff and back in.

closed 2017-01-25 by lordofhyphens

gege2b, 2017-01-25:
as mentionned earlier, removing the perl5 directory (and flushing my slic3r local repo) made it work again 
And I never ran into this problem again since then
thanks :)

lordofhyphens, 2017-01-26:
Glad to hear it.