slic3r/Slic3r #3625: Object settings crashes slic3r

Tinchus2009, 2016-12-18:
### Version
1.2.9-568-g807d441 (latest pull)

### Operating system type + version
win10 64

### Behavior
Im trying the new features, the slice now command works perfectly nice. Then I tried to open the setting for my model. I tried double clicking method and right click > settings, for the model, both cases make slicer crash.  I get "Perl interpreter stopped working"

I tried different models, Y tried loading different slicing profiles. But always the same: opening the object settings crashes slicer.

How can I give you mode info? I dont get any log bisdes the message of the perl interpreter stopping working.

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

alexrj, 2016-12-18:
Did you issue the Build.PL command after pulling?

Tinchus2009, 2016-12-18:
yes: perl build.pl and then perld build.pl --gui

alexrj, 2016-12-19:
I can't reproduce this on OS X using current HEAD.

Tinchus2009, 2016-12-19:
Well... OSX and win 10, the only thing in common is just the number 10 :(

lordofhyphens, 2016-12-31:
It's probably the 3D preview and some stupid garbage involving wxPerl and/or Windows. Again.

lordofhyphens, 2017-01-01:
It's 99% the 3d preview, as it crashes when you try to open the Cut dialog.

lordofhyphens, 2017-01-02:
The crash point for the object settings is at the ShowModal call at Plater.pm, line 1715. That's a call to a Wx function which actually does the drawing (shows the dialog).

To dig deeper I'll probably need to build WxWidgets library through Alien (sigh) with debug flag on.

lordofhyphens, 2017-01-02:
For Cut it looks like it just choked on a part with multiple shells, will post a separate issue.

lordofhyphens, 2017-01-02:
accd174566ae9fcc77b662d22ab3a93c43b73add is the first bad commit ;D

lordofhyphens, 2017-01-02:
Reverting that commit (and fixing the merge issue because there was a later tweak) avoids the crash. Currently looking at what broke it on Windows.

lordofhyphens, 2017-01-02:
`$self->{move_target}` had its distance_to function being called before it was assigned. Apparently perl can handle this on OSX and Linux but not on Windows (at least Strawberry Perl). 

Fixed via ed0173364f56ce0b508e3f19e6e6bdebf71cdf77

closed 2017-01-02 by lordofhyphens