slic3r/Slic3r #3628: Feature Request: Alternative cooling strategy: Sit and Wait

Coolway99, 2016-12-19:
### Version
Whatever's next, preferably 1.3.0

### Operating system type + version
Windows 7

### Behavior
As someone who prints in ABS, I hear it time and time again to NOT use a printing fan. This isn't supported to well by Slic3r, who can only slow down and/or use a fan. I think it wouldn't be difficult to add a third strategy to this: sit and wait.

Using a pre-defined point, or better yet have a visible cylinder on the build plate to use (doesn't actually print though, just for reference), you can have it park there until the build time is greater than the specified layer time. This prevents overheating on ABS without using a cooling fan. It would be simple to do thanks to G4- "Dwell". G4 P10000 would be 10 seconds.

_Is this a new feature request?_ I hope so, wasn't able to find something like it

#### STL/Config (.ZIP) where problem occurs
N/A

Note: I am unsure if this was already implemented or not. If so, sorry for your time, I was unable to build slic3r on my own comp and verify this for myself

lordofhyphens, 2016-12-20:
I think if you set max speed to 0 for the fan it will properly slow down.

There's some extra perils involved though with ABS and dwelling, as you'll
almost certainly lose pressure, especially as you will have to dwell away
from the print.

FWIW my recently added lambda code could be modified to add generic cooling
towers (that is, a small cylinder used to get the hotend printing something
else over *there*), but I'm not sure that's the best option. In my original
cuts of the code I had added an "Add Generic" button to the main section.
Perhaps a cooling tower option could be planted in some legal part of the
bed that doesn't have overlaps and it could be told to dwell over there
(where we don't care what happens). Print it with sparse infill to make
sure that we don't waste a lot of filament but it is still relatively
stable so it doesn't fall over on its own.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2016-12-20:
There *should* should be a working build on bintray.com (see the Readme). I
will take a look at it to see if recent commits have broken it or its
packaging (hint: they probably have, although things should be much much
easier now to package)

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

Coolway99, 2016-12-20:
Perhaps this would be a little more complex to implement, but it's still food for though. Treat it as support material up until the part where it actually needs to start lining up with the layers. For example, 8 small pillars but when it gets up to an area that needs to be cooled via waiting then change to say, printing an octagon. Since we don't care how the cooling part turns out, then we can afford to bridge something that would obviously look horrid without proper supports.

I have a couple other ideas, but another major one is that it could be offloaded onto any current support material on the print. For example, if the next layer is support, include it in the cooling time calculation. It won't always work, especially where support material doesn't like up perfectly with the print (in my case where my support is 0.2mm and I use layers of 0.152mm), but perhaps with a few tweaks it could work.

lordofhyphens, 2016-12-20:
The important thing here is to clearly identify the problem, note why it's
a problem ofc. Cooling towers is just one possible solution to the problem.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2017-01-25:
This used to be called "dwell" or "orbit" in Skeinforge and was intentionally left out, back in the day.