slic3r/Slic3r #3630: multiple bridging zones should be merged

djdelorie, 2016-12-20:
### Version
1.2.9-579-g12165c7
### Operating system type + version
Fedora 24

### Behavior

When multiple reasons for bridging exist in the same layer, multiple bridging "zones" are created which cause bridges to not span to existing edges.  These zones should be merged so that bridging is successful.

[image]
[bridge-bug2.zip](https://github.com/alexrj/Slic3r/files/664398/bridge-bug2.zip)

lordofhyphens, 2017-01-25:
Probably a duplicate of #3578. If your test case doesn't fail in the Prusa3D fork, I can close this as a duplicate (and go looking around in the Prusa3D fork to bring over that bridge merge code, although it probably needs good test cases too).

bubnikv, 2017-01-25:
This is not a good example, the "bridges" are not bridges at all, they have
no anchor to connect to.

djdelorie, 2017-01-25:
bubnikv, are you saying that my example isn't a good example of bridging, isn't a good example of the problem in 3578, or that 3578 isn't a good example of the problem I describe here?

djdelorie, 2017-01-25:
still broken, in a slightly different way, with the prusa fork...  just incrementally increase top layers until you get this:
http://www.delorie.com/photos/3d_prints/prusa-slic3r-bridging.html

bubnikv, 2017-01-27:
> bubnikv, are you saying that my example isn't a good example of bridging, isn't a good example of the problem in 3578, or that 3578 isn't a good example of the problem I describe here?

I think I am starting to understand the misunderstanding.

You provide a closed object to Slic3r. You disabled infills, and you hope that Slic3r will consider it to be a hollow solid. But this is not the case. For the ceiling to be bridged correctly, you have to provide a hollow solid to Slic3r.

The explanation is following. Slic3r creates bridges by turning infill into bridging infill. Also Slic3r turns unsupported parts of the perimeters into bridging perimeters if allowed. To correctly anchor a bridge, the infill edges have to be placed over some support. In your example, this would require Slic3r to reduce the inner perimeter to create a step, over which the infill would be anchored. This happens if you make the object body hollow, but not if you just set the infill to zero.

I am not sure what to do about your request. All in all, it may make sense to allow Slic3r to anchor the roofs of hollow objects into perimeters to save material on infills, but this may have a negative effect on the quality of the outer walls of the print. So we would likely end up with another switch in Slic3r, which is not desirable in general case.

lordofhyphens, 2017-04-03:
@djdelorie Please retest with current master, there's been some changes to bridging logic.

djdelorie, 2017-04-04:
No change, sorry.