slic3r/Slic3r #3631: Test errors in latest pull build

Tinchus2009, 2016-12-21:
### Version
Latest Pull: 6d3c0e3e98edabd58a2cc4c1d2fc9ff676ea468d

### Operating system type + version
win10 64b

### Behavior

Build reports this errors:
Installing C:\Slic3r\LOCAL-~1\lib\perl5\MSWin32-x64-multi-thread\auto\Slic3r\XS\XS.dll
OK
Successfully installed Slic3r-XS-0.01
1 distribution installed
t\angles.t ..................... ok
t\arcs.t ....................... ok
t\avoid_crossing_perimeters.t .. ok
t\bridges.t .................... ok
t\clean_polylines.t ............ ok
t\clipper.t .................... ok
t\collinear.t .................. ok
t\combineinfill.t .............. ok
t\config.t ..................... ok
t\cooling.t .................... 1/12 No layers were detected. You might want to repair your STL file(s) or check their size or thickness and retry.
# Looks like your test exited with 255 just after 8.
t\cooling.t .................... Dubious, test returned 255 (wstat 65280, 0xff00)
Failed 4/12 subtests
t\custom_gcode.t ............... 11/15 No layers were detected. You might want to repair your STL file(s) or check their size or thickness and retry.
# Looks like your test exited with 255 just after 13.
t\custom_gcode.t ............... Dubious, test returned 255 (wstat 65280, 0xff00)
Failed 2/15 subtests
t\dynamic.t .................... skipped: variable-width paths are currently disabled
t\fill.t ....................... ok
t\flow.t ....................... 1/6 No layers were detected. You might want to repair your STL file(s) or check their size or thickness and retry.
# Looks like your test exited with 255 just after 1.
t\flow.t ....................... Dubious, test returned 255 (wstat 65280, 0xff00)
Failed 5/6 subtests
t\gaps.t ....................... ok
t\gcode.t ...................... ok
t\geometry.t ................... ok
t\layers.t ..................... ok
t\loops.t ...................... skipped: temporarily disabled
t\multi.t ...................... ok
t\perimeters.t ................. 1/59 No layers were detected. You might want to repair your STL file(s) or check their size or thickness and retry.
# Looks like your test exited with 255 just after 44.
t\perimeters.t ................. Dubious, test returned 255 (wstat 65280, 0xff00)
Failed 15/59 subtests
t\polyclip.t ................... ok
t\pressure.t ................... ok
t\print.t ...................... ok
t\retraction.t ................. ok
t\shells.t ..................... ok
t\skirt_brim.t ................. 2/6 No layers were detected. You might want to repair your STL file(s) or check their size or thickness and retry.
# Looks like your test exited with 255 just after 4.
t\skirt_brim.t ................. Dubious, test returned 255 (wstat 65280, 0xff00)
Failed 2/6 subtests
t\slice.t ...................... skipped: temporarily disabled
t\support.t .................... 1/27 No layers were detected. You might want to repair your STL file(s) or check their size or thickness and retry.
# Looks like your test exited with 255 just after 12.
t\support.t .................... Dubious, test returned 255 (wstat 65280, 0xff00)
Failed 15/27 subtests
t\svg.t ........................ ok
t\thin.t ....................... ok
t\threads.t .................... ok
t\vibrationlimit.t ............. ok

Test Summary Report
-------------------
t\cooling.t                  (Wstat: 65280 Tests: 8 Failed: 0)
  Non-zero exit status: 255
  Parse errors: Bad plan.  You planned 12 tests but ran 8.
t\custom_gcode.t             (Wstat: 65280 Tests: 13 Failed: 0)
  Non-zero exit status: 255
  Parse errors: Bad plan.  You planned 15 tests but ran 13.
t\flow.t                     (Wstat: 65280 Tests: 1 Failed: 0)
  Non-zero exit status: 255
  Parse errors: Bad plan.  You planned 6 tests but ran 1.
t\perimeters.t               (Wstat: 65280 Tests: 44 Failed: 0)
  Non-zero exit status: 255
  Parse errors: Bad plan.  You planned 59 tests but ran 44.
t\skirt_brim.t               (Wstat: 65280 Tests: 4 Failed: 0)
  Non-zero exit status: 255
  Parse errors: Bad plan.  You planned 6 tests but ran 4.
t\support.t                  (Wstat: 65280 Tests: 12 Failed: 0)
  Non-zero exit status: 255
  Parse errors: Bad plan.  You planned 27 tests but ran 12.
Files=33, Tests=477, 69 wallclock secs ( 0.14 usr +  0.06 sys =  0.20 CPU)
Result: FAIL
Some tests failed. Please report the failure to the author!

Anyway, it seems slic3r was built, but launching it gives some strange behaviour: I loaded some models, all of them being centered in the plater, at the moment of slicing them, the sliced model is relocated to 0,0 coordinates, the venter of the model, so half model ends up outside the plater

Tinchus2009, 2016-12-21:
All this error have desapeared in the latest 1.2.9-593-g4a25b3e

closed 2016-12-21 by Tinchus2009

alexrj, 2016-12-21:
Thank you for reporting.