slic3r/Slic3r #3632: [1.3.0-dev] Wrong object placement in ported _slice_region()

xoan, 2016-12-21:
### Version
32fdd4d

### Operating system type + version
Arch Linux

### Behavior
Both _3D/2D_ views in plater shows the correct position of objects in the bed plate, but _Preview/Layers_ tabs doesn't.

[image]
[image]
[image]
[image]

alexrj, 2016-12-21:
Sorry.  Shoud be fixed now.

xoan, 2016-12-21:
Please, don't say _sorry_ ;)

closed 2016-12-21 by xoan

alexrj, 2016-12-21:
:-))))