slic3r/Slic3r #3635: [1.3.0-dev] Build.pl creates ~/perl5 directory even if it is not used

xoan, 2016-12-22:
### Version
Current master: 117fb30

### Operating system type + version
Arch Linux

### Behavior
``` zsh
% env | grep PERL
PERL5LIB=/home/xoan/.perl/lib/perl5
PERL_LOCAL_LIB_ROOT=/home/xoan/.perl
PERL_MB_OPT=--install_base "/home/xoan/.perl"
PERL_MM_OPT=INSTALL_BASE=/home/xoan/.perl
```
``` zsh
% perl Build.PL
/home/xoan/.perl/bin/cpanm
Attempting to create directory /home/xoan/perl5
App::cpanminus is up to date. (1.7042)
Class::XSAccessor is up to date. (1.19)
...
XML::SAX::ExpatXS is up to date. (1.33)
threads is up to date. (2.09)
cpanm (App::cpanminus) 1.7042 on perl 5.024000 built for x86_64-linux-thread-multi
```

lordofhyphens, 2016-12-22:
That is probably the local::lib install. I will see if it can be squelched.

xoan, 2016-12-22:
This is the line which creates the ~/perl5 directory:

```perl
if (!eval "use local::lib; 1") {
```

I have [local::lib](https://www.archlinux.org/packages/community/any/perl-local-lib/) installed by package manager in Arch Linux.

alexrj, 2016-12-22:
Oh right, I know how to fix it, I'll do right now.

alexrj, 2016-12-22:
Done!

closed 2016-12-22 by alexrj