slic3r/Slic3r #3636: Slic3r 1.3.0 dev unhandled exception with support enabled

jkoljo, 2016-12-22:
### Version
Latest pull of Slic3r dev, compiled according to Windows system instructions, strawberry perl
Latest commit 117fb30 

### Operating system type + version
Windows 7

### Behavior
Slic3r throws unhandled exception if support is enabled and I go to preview or 3D tabs. If support is not enabled, slicing seems to work fine.

Steps to reproduce:
* Load model (maybe any model, but one example attached)
* Enable support
* Allow it to slice
* Go to preview tab
* -> Exception

#### STL/Config (.ZIP) where problem occurs

[problem.zip](https://github.com/alexrj/Slic3r/files/668436/problem.zip)

xoan, 2016-12-22:
Using your config and your model in Arch Linux:

```
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
```

The only difference between _works_ and _fails_ config is that the last one have support material enabled, right?

I've tried activating support in my config ([FoldaRap.ini.zip](https://github.com/alexrj/Slic3r/files/668467/FoldaRap.ini.zip)) and Slic3r does not crash with your model. Maybe it could helps to debugging...

jkoljo, 2016-12-22:
Yep that is the only difference. Thanks for the config file, I will play with it when I can :)

alexrj, 2016-12-22:
Thank you for this report, I just fixed it.

closed 2016-12-22 by alexrj