slic3r/Slic3r #3637: "Only infill where needed" crashes slic3r

Patola, 2016-12-22:
### Version
_Version of Slic3r used goes here_
1.3.0-dev, latest one from git as of 12/22/2016 - 1.2.9-606-g73763a8, HEAD 73763a8417751b6f45e71cebad6bf4a8a8827993

### Operating system type + version
Ubuntu 16.04 64 bits with latest updates

### Behavior
* _Describe the problem_
Setting the "only infill where needed" option makes slic3r segfault. It starts slicing but at some point just crashes and disappears.

* _Steps needed to reproduce the problem_

Open slic3r, load any STL (tried half a dozen STLs and it always happens), go to print settings → infill → reducing printing time → then set [x] Only infill where needed.

* _Expected Results_

Slicing is finished.

* _Actual Results_

Slic3r crashes with a segfault and these console messages:
```
[16:08] [3150] [patola@ovo Slic3r]% ./slic3r.pl
*** Error in `perl': double free or corruption (fasttop): 0x00007fd300113120 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7fd33bde17e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x7fe0a)[0x7fd33bde9e0a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7fd33bded98c]
/dados/git/Slic3r/local-lib/lib/perl5/x86_64-linux-gnu-thread-multi/auto/Slic3r/XS/XS.so(_ZNSt8_Rb_treeIlSt4pairIKlN6Slic3r15FillRectilinear17IntersectionPointEESt10_Select1stIS5_ESt4lessIlESaIS5_EE12_M_erase_auxESt23_Rb_tree_const_iteratorIS5_E+0x26)[0x7fd33a496296]
/dados/git/Slic3r/local-lib/lib/perl5/x86_64-linux-gnu-thread-multi/auto/Slic3r/XS/XS.so(_ZN6Slic3r15FillRectilinear22_fill_single_directionENS_9ExPolygonERKSt4pairIfNS_5PointEElPSt6vectorINS_8PolylineESaIS8_EE+0x1b64)[0x7fd33a4915a4]
/dados/git/Slic3r/local-lib/lib/perl5/x86_64-linux-gnu-thread-multi/auto/Slic3r/XS/XS.so(_ZN6Slic3r15FillRectilinear20_fill_surface_singleEjRKSt4pairIfNS_5PointEERNS_9ExPolygonEPSt6vectorINS_8PolylineESaIS9_EE+0x273)[0x7fd33a492c93]
/dados/git/Slic3r/local-lib/lib/perl5/x86_64-linux-gnu-thread-multi/auto/Slic3r/XS/XS.so(_ZN6Slic3r4Fill12fill_surfaceERKNS_7SurfaceE+0x1ce)[0x7fd33a53bd1e]
/dados/git/Slic3r/local-lib/lib/perl5/x86_64-linux-gnu-thread-multi/auto/Slic3r/XS/XS.so(_ZN6Slic3r11LayerRegion9make_fillEv+0x1351)[0x7fd33a45c6a1]
/dados/git/Slic3r/local-lib/lib/perl5/x86_64-linux-gnu-thread-multi/auto/Slic3r/XS/XS.so(_ZN6Slic3r5Layer10make_fillsEv+0x24)[0x7fd33a465454]
/dados/git/Slic3r/local-lib/lib/perl5/x86_64-linux-gnu-thread-multi/auto/Slic3r/XS/XS.so(_ZN6Slic3r15_parallelize_doIPNS_5LayerEEEvPSt5queueIT_St5dequeIS4_SaIS4_EEEPN5boost5mutexENSA_8functionIFvS4_EEE+0x97)[0x7fd33a538d77]
/dados/git/Slic3r/local-lib/lib/perl5/x86_64-linux-gnu-thread-multi/auto/Slic3r/XS/XS.so(_ZN5boost6detail11thread_dataINS_3_bi6bind_tIvPFvPSt5queueIPN6Slic3r5LayerESt5dequeIS7_SaIS7_EEEPNS_5mutexENS_8functionIFvS7_EEEENS2_5list3INS2_5valueISC_EENSL_ISE_EENSL_ISH_EEEEEEE3runEv+0x48)[0x7fd33a539358]
/usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0(+0x115d5)[0x7fd339a845d5]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x76ba)[0x7fd33c13a6ba]
/lib/x86_64-linux-gnu/libc.so.6(clone+0x6d)[0x7fd33be7082d]
======= Memory map: ========
00400000-005cf000 r-xp 00000000 fc:01 138466                             /usr/bin/perl
007ce000-007cf000 r--p 001ce000 fc:01 138466                             /usr/bin/perl
007cf000-007d2000 rw-p 001cf000 fc:01 138466                             /usr/bin/perl
0143d000-0c0f9000 rw-p 00000000 00:00 0                                  [heap]
404a0000-4053c000 rw-p 00000000 00:00 0 
40bfe000-40c00000 r-xs 00000000 fc:03 295                                /tmp/.glYWFaJ9 (deleted)
41621000-416bd000 rw-p 00000000 00:00 0 
41ae1000-41b7d000 rw-p 00000000 00:00 0 
7fd2d9a1e000-7fd2db536000 rw-s 2580e5000 00:06 639                       /dev/nvidiactl
7fd2ddfff000-7fd2e0000000 rw-p 00000000 00:00 0 
7fd2e0000000-7fd2e023d000 rw-p 00000000 00:00 0 
7fd2e023d000-7fd2e4000000 ---p 00000000 00:00 0 
7fd2e5fff000-7fd2e8000000 rw-p 00000000 00:00 0 
7fd2e8000000-7fd2e814b000 rw-p 00000000 00:00 0 
7fd2e814b000-7fd2ec000000 ---p 00000000 00:00 0 
7fd2ec000000-7fd2ec1fe000 rw-p 00000000 00:00 0 
7fd2ec1fe000-7fd2f0000000 ---p 00000000 00:00 0 
7fd2f0000000-7fd2f02a6000 rw-p 00000000 00:00 0 
7fd2f02a6000-7fd2f4000000 ---p 00000000 00:00 0 
7fd2f4000000-7fd2f4545000 rw-p 00000000 00:00 0 
7fd2f4545000-7fd2f8000000 ---p 00000000 00:00 0 
7fd2f8000000-7fd2f81fe000 rw-p 00000000 00:00 0 
7fd2f81fe000-7fd2fc000000 ---p 00000000 00:00 0 
7fd2fd2d1000-7fd2fd2d2000 ---p 00000000 00:00 0 
7fd2fd2d2000-7fd2fdad2000 rw-p 00000000 00:00 0 
7fd2fdad2000-7fd2fdef1000 rw-s 218d35000 00:06 639                       /dev/nvidiactl
7fd2fe6bf000-7fd2feff6000 rw-s 213610000 00:06 639                       /dev/nvidiactl
7fd2feff6000-7fd300000000 rw-s 2744f5000 00:06 639                       /dev/nvidiactl
7fd300000000-7fd30024e000 rw-p 00000000 00:00 0 
[... 145 lines cut ...]
7fd324eac000-7fd324ead000 r--p 00003000 fc:01 589228                     /usr/lib/x86_64-linux-gnu/perl/5.22.1/auto/IO/IO.so
7fd324ead000-7fd324eae000 rw-p 00004000 fc:01 589228                     /usr/lib/x86_64-linux-gnu/perl/5.22.1/auto/IO/IO.so
7fd324eae000-7fd324eb4000 r-xp 00000000 fc:01 589210                     /usr/lib/x86_64-linux-gnu/perl/5.22.1/auto/File/Glob/Glob.so
7fd324eb4000-7fd3250b3000 ---p 00006000 fc:01 589210                     /usr/lib/x86_64-linux-gnu/perl/5.22.1/auto/File/Glob/Glob.so
7fd3250b3000-7fd3250b4000 r--p 00005000 fc:01 589210                     /usr/lib/x86_64-linux-gnu/perl/5.22.1/auto/File/Glob/Glob.so
7fd3250b4000-7fd3250b5000 rw-p 00006000 fc:01 589210                     /usr/lib/x86_64-linux-gnu/perl/5.22.1/auto/File/Glob/Glob.so
7fd3250b5000-7fd325106000 r-xp 00000000 fc:01 656004                     /usr/local/lib/x86_64-linux-gnu/perl/5.22.1/auto/Wx/Grid/Grid.so
7fd325106000-7fd325305000 ---p 00051000 fc:01 656004                     /usr/local/lib/x86_64-linux-gnu/perl/5.22.1/auto/Wx/Grid/Grid.so
7fd325305000-7fd325307000 r--p 00050000 fc:01 656004                     /usr/local/lib/x86_64-linux-gnu/perl/5.22.1/auto/Wx/Grid/Grid.so
7fd325307000-7fd325308000 rw-p 00052000 fc:01 656004                     /usr/local/lib/x86_64-linux-gnu/perl/5.22.1/auto/Wx/Grid/Grid.so
7fd325308000-7fd32531b000 r-xp 00000000 fc:01 142853                     /usr/lib/x86_64-linux-gnu/libwx_gtk2u_gl-3.0.so.0.2.0
7fd32531b000-7fd32551b000 ---p 00013000 fc:01 142853                     /usr/lib/x86_64-linux-gnu/libwx_gtk2u_gl-3.0.so.0.2.0
7fd32551b000-7fd32551d000 r--p 00013000 fc:01 142853                     /usr/lib/x86_64-linux-gnu/libwx_gtk2u_gl-3.0.so.0.2.0
7fd32551d000-7fd32551e000 rw-p 00015000 fc:01 142853                     /usr/lib/x86_64-linux-gnu/libwx_gtk2u_gl-3.0.so.0.2.0
7fd32551e000-7fd325524000 r-xp 00000000 fc:01 658579                     /usr/local/lib/x86_64-linux-gnu/perl/5.22.1/auto/Wx/GLCanvas/GLCanvas.sozsh: abort (core dumped)  ./slic3r.pl

```

  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
[Nuke_Lamp.zip](https://github.com/alexrj/Slic3r/files/669488/Nuke_Lamp.zip)

Tinchus2009, 2016-12-22:
I have built this morning vrsion 1.2.9-606-g73763a8 under win 10 64b, and I have not been able to reproduce the problem. I tried slicing 3 models, activateing that option and they sliced perfectly (nevertheless it crashes when I try to open "object settings"

Tinchus2009, 2017-01-01:
Update: In my previous reports I confirmed the issue in my win 10 system. Nevertheless, the crash happens ONLY if there is 2 or more objects in the plate. If I slice only 1 object, there is no crash. I dont upload examples because I have tried  many models and the same happens with all of them and with different configuration also.

Tinchus2009, 2017-01-01:
Update: my first comment was done based on a first try when I read the report. But now, I have tested it further and the problem is ramdomly happening. With the same object/s I have sometimes a crash and other times the object is sliced with no problem. Still cant figure it out what changes between slices....

lordofhyphens, 2017-01-25:
Closing as this appears to be the same issue as in #3648 and that has more commentary present.

closed 2017-01-25 by lordofhyphens