slic3r/Slic3r #3638: First layer travel speed

claell, 2016-12-25:
There does not seem to be an option to set a special travel speed for the first layer.
Currently I have set a travel speed of 150mm/s and I fear that it is dragging the filament from the bottom at this speed like described in https://ultimaker.com/en/community/8316-suggestion-bottom-layer-travel-speed

I am pretty new to 3d printing and have adhesion problems unrelated to the travel speed, so I am not sure, whether this is really needed.

alexrj, 2016-12-25:
Travel speed should just be the fastest your machine can do. There's not advantage in slower travel speed.

claell, 2016-12-25:
So you say the problem with dragged away filament is not there or caused by a bad machine setup?

lordofhyphens, 2016-12-25:
Usually, yes.

claell, 2016-12-25:
Alright. So then I will close this.

closed 2016-12-25 by claell

alexrj, 2016-12-25:
Dragged away filament means bad adhesion. Even if you somehow managed to prevent it from being dragged away, you'd have other issues related to bad adhesion. So raise your temperatures, clean build plate etc.