slic3r/Slic3r #3640: Segfault inside Perl / WxWidgets Events

jonas-schievink, 2016-12-28:
### Version

1.2.9

### Operating system type + version

Arch Linux x86_64

### Behavior

A segfault occurs somewhere inside Perl / WxWidgets Event Handling. I was playing around with different support material settings and patterns. This happens spuriously and is a bit hard to reproduce, it's also independent of the STL loaded (I think). Maybe it doesn't even have anything to do with supports.

Happened while trying to print http://www.thingiverse.com/thing:1776429.

```
Thread 1 "slic3r" received signal SIGSEGV, Segmentation fault.
0x00007ffff7b15ab6 in Perl_pp_entersub () from /opt/slic3r/bin/../dll/libperl.so
(gdb) bt
#0  0x00007ffff7b15ab6 in Perl_pp_entersub () at /opt/slic3r/bin/../dll/libperl.so
#1  0x00007ffff7aae230 in Perl_call_sv () at /opt/slic3r/bin/../dll/libperl.so
#2  0x00007ffff7af8e0c in  () at /opt/slic3r/bin/../dll/libperl.so
#3  0x00007ffff7afa9b9 in Perl_vwarn () at /opt/slic3r/bin/../dll/libperl.so
#4  0x00007ffff7afab93 in Perl_warner () at /opt/slic3r/bin/../dll/libperl.so
#5  0x00007ffff7b46f88 in Perl_free_tmps () at /opt/slic3r/bin/../dll/libperl.so
#6  0x00007ffff7ab2bd2 in perl_destruct () at /opt/slic3r/bin/../dll/libperl.so
#7  0x00007ffff55ff272 in  () at /opt/slic3r/lib/vrt/5896808d50a83c9caf32aebcd5db660e/threads.so
#8  0x00007ffff5600a5b in  () at /opt/slic3r/lib/vrt/5896808d50a83c9caf32aebcd5db660e/threads.so
#9  0x00007ffff7b15db3 in Perl_pp_entersub () at /opt/slic3r/bin/../dll/libperl.so
#10 0x00007ffff7b143fe in Perl_runops_standard () at /opt/slic3r/bin/../dll/libperl.so
#11 0x00007ffff7aae28b in Perl_call_sv () at /opt/slic3r/bin/../dll/libperl.so
#12 0x00007ffff358758c in wxPliEventCallback::Handler(wxEvent&) () at /opt/slic3r/lib/vrt/8b57d08e0d55cfe6498642b59ecf5ad1/Wx.so
#13 0x00007ffff2463df0 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) () at /opt/slic3r/bin/../dll/libwx_baseu-2.9.so.3
#14 0x00007ffff2463e79 in wxEvtHandler::SearchDynamicEventTable(wxEvent&) () at /opt/slic3r/bin/../dll/libwx_baseu-2.9.so.3
#15 0x00007ffff2463ff8 in wxEvtHandler::TryHereOnly(wxEvent&) () at /opt/slic3r/bin/../dll/libwx_baseu-2.9.so.3
#16 0x00007ffff246491d in wxEvtHandler::ProcessEventLocally(wxEvent&) () at /opt/slic3r/bin/../dll/libwx_baseu-2.9.so.3
#17 0x00007ffff2464974 in wxEvtHandler::ProcessEvent(wxEvent&) () at /opt/slic3r/bin/../dll/libwx_baseu-2.9.so.3
#18 0x00007ffff2464426 in wxEvtHandler::ProcessPendingEvents() () at /opt/slic3r/bin/../dll/libwx_baseu-2.9.so.3
#19 0x00007ffff22ec9f7 in wxAppConsoleBase::ProcessPendingEvents() () at /opt/slic3r/bin/../dll/libwx_baseu-2.9.so.3
#20 0x00007ffff29491de in wxApp::DoIdle() () at /opt/slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3
#21 0x00007ffff29492a3 in  () at /opt/slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3
#22 0x00007ffff049643a in g_main_context_dispatch () at /usr/lib/libglib-2.0.so.0
#23 0x00007ffff04967f0 in  () at /usr/lib/libglib-2.0.so.0
#24 0x00007ffff0496b12 in g_main_loop_run () at /usr/lib/libglib-2.0.so.0
#25 0x00007ffff1d6e3a7 in gtk_main () at /usr/lib/libgtk-x11-2.0.so.0
#26 0x00007ffff2966f7b in wxGUIEventLoop::Run() () at /opt/slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3
#27 0x00007ffff22ecca6 in wxAppConsoleBase::MainLoop() () at /opt/slic3r/bin/../dll/libwx_baseu-2.9.so.3
#28 0x00007ffff349c031 in wxPliApp::MainLoop() () at /opt/slic3r/lib/vrt/8b57d08e0d55cfe6498642b59ecf5ad1/Wx.so
#29 0x00007ffff345a3c2 in  () at /opt/slic3r/lib/vrt/8b57d08e0d55cfe6498642b59ecf5ad1/Wx.so
#30 0x00007ffff7b15db3 in Perl_pp_entersub () at /opt/slic3r/bin/../dll/libperl.so
#31 0x00007ffff7b143fe in Perl_runops_standard () at /opt/slic3r/bin/../dll/libperl.so
#32 0x00007ffff7aaeba3 in perl_run () at /opt/slic3r/bin/../dll/libperl.so
#33 0x0000000000403064 in  ()
#34 0x00007ffff6972291 in __libc_start_main () at /usr/lib/libc.so.6
#35 0x0000000000401b69 in  ()
#36 0x00007fffffffe408 in  ()
#37 0x000000000000001c in  ()
#38 0x0000000000000001 in  ()
#39 0x00007fffffffe768 in  ()
#40 0x0000000000000000 in  ()
```

#### STL/Config (.ZIP) where problem occurs

Due to the segfault, I've lost the original plate and config, but this should be close (the objects are in a different order, config should match closely).

[config-and-plate.zip](https://github.com/alexrj/Slic3r/files/675944/config-and-plate.zip)

lordofhyphens, 2016-12-28:
Can you reproduce on 1.3.0-dev?

jonas-schievink, 2016-12-29:
Not yet, I'll use it for a bit

jonas-schievink, 2016-12-29:
Possibly related (on current master):

```
==21099== Use of uninitialised value of size 8
==21099==    at 0x4F097F5: Perl_pp_and (in /usr/lib/perl5/core_perl/CORE/libperl.so)
==21099==    by 0x4F09295: Perl_runops_standard (in /usr/lib/perl5/core_perl/CORE/libperl.so)
==21099==    by 0x4E897CD: Perl_call_sv (in /usr/lib/perl5/core_perl/CORE/libperl.so)
==21099==    by 0xBCC73FC: wxPliEventCallback::Handler(wxEvent&) (in /home/jonas/dev/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Wx/Wx.so)
==21099==    by 0xCD2C43D: wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const (in /usr/lib/libwx_baseu-3.0.so.0.2.0)
==21099==    by 0xCEB2A76: wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) (in /usr/lib/libwx_baseu-3.0.so.0.2.0)
==21099==    by 0xCEB2E5D: wxEvtHandler::SearchDynamicEventTable(wxEvent&) (in /usr/lib/libwx_baseu-3.0.so.0.2.0)
==21099==    by 0xCEB2EED: wxEvtHandler::TryHereOnly(wxEvent&) (in /usr/lib/libwx_baseu-3.0.so.0.2.0)
==21099==    by 0xCEB2FB2: wxEvtHandler::ProcessEventLocally(wxEvent&) (in /usr/lib/libwx_baseu-3.0.so.0.2.0)
==21099==    by 0xCEB3014: wxEvtHandler::ProcessEvent(wxEvent&) (in /usr/lib/libwx_baseu-3.0.so.0.2.0)
==21099==    by 0xCEB2D76: wxEvtHandler::SafelyProcessEvent(wxEvent&) (in /usr/lib/libwx_baseu-3.0.so.0.2.0)
==21099==    by 0x19C93397: wxGLCanvas::OnInternalIdle() (in /usr/lib/libwx_gtk2u_gl-3.0.so.0.2.0)
```

jonas-schievink, 2016-12-29:
Probably unrelated, but Valgrind also reported this invalid free (also on current master):

```
==23125== Invalid free() / delete / delete[] / realloc()
==23125==    at 0x4C2BD3A: free (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==23125==    by 0x530C2AF: tdestroy_recurse (in /usr/lib/libc-2.24.so)
==23125==    by 0x53703EB: free_mem (in /usr/lib/libc-2.24.so)
==23125==    by 0x5370542: __libc_freeres (in /usr/lib/libc-2.24.so)
==23125==    by 0x4A2573F: _vgnU_freeres (in /usr/lib/valgrind/vgpreload_core-amd64-linux.so)
==23125==    by 0x525C942: __run_exit_handlers (in /usr/lib/libc-2.24.so)
==23125==    by 0x525C9E9: exit (in /usr/lib/libc-2.24.so)
==23125==    by 0x400D4A: main (in /usr/bin/perl)
==23125==  Address 0xba52dc0 is 624 bytes inside an unallocated block of size 656 in arena "client"
```

jonas-schievink, 2016-12-31:
Haven't managed to reproduce the original crash on master. Closing.

closed 2016-12-31 by jonas-schievink