slic3r/Slic3r #3641: Errors when slicing 3D Benchy

brrmatteo, 2016-12-29:
### Version
_1.2.9 x64 win_

### Operating system type + version
Windows 7 Ultimate 64bit 

Hello, when i try to slice 3DBenchy (.stl attached), in "Preview" i got a huge error (.jpg attached) of slicing and the result is a print with this error.

### Error screenshot:
[image]

### How model should be:
[image]

### Config file:
[config.zip](https://github.com/alexrj/Slic3r/files/677923/config.zip)

### .stl file
[3DBenchy_-_Multi-part_-_Complete_17_shells_-_3DBenchy.com.zip](https://github.com/alexrj/Slic3r/files/677918/3DBenchy_-_Multi-part_-_Complete_17_shells_-_3DBenchy.com.zip)

lordofhyphens, 2016-12-29:
Yeah that model has 522557 "repaired" errors for manifoldness. Ergo, trust
nothing about it to work. At all.

The basic 3DBenchy.stl works fine on 1.3.0-dev.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2016-12-29:
If you see issues with the model and it's being reported as manifold, post and I'll reopen.

closed 2016-12-29 by lordofhyphens