slic3r/Slic3r #3642: Saving Object settings & amf issue

fobetor, 2016-12-30:
Hi
 I have a doubt that may be easy, but.. I cant find the answer.
Is ther a way to save an object settings? I mean, I take the time to prepare settings & modifiers for an Object, and... I cant find where I can save all for the future. Do I need to redo every setting & Modiyer each time I close & open Slic3r?
OR , I can save the object in some format that maintainsd all the work done so I can open Slic3r and load the file and have everything ready to go?

Now, An Issue I have found is that I cant read amf files saved from Repetier or Netfabb (freeversion). Slic3r gives me an error.
Thanks 
 Fernando
### Version
1.2.9

lordofhyphens, 2016-12-30:
No, but I believe that @bubnikv is working on it.

fobetor, 2016-12-30:
That would be a great new feature ;-)

lordofhyphens, 2017-01-25:
For anything involving repetierhost, we can't support that use case at all. RepetierHost is a closed source program, contact them for support about that.

If you can reproduce the problem with Slic3r 1.3.0-dev on its own, then we can add a separate issue for that (please keep separate issues separate). Going to need a test case in that regard

lordofhyphens, 2017-01-25:
Primary issue is duplicate of #3545.

If you are still having problems with the second reported issue and can reproduce in 1.3.0-dev (see README.md for a windows link), please create a separate issue with a test model and config (as the guidelines indicate).

Closing.

closed 2017-01-25 by lordofhyphens