slic3r/Slic3r #3643: Consider moving retraction settings to Filament Settings

jammi, 2017-01-01:
Usability-wise, I'd like the retraction length setting to be a part of the filament profile, since different filaments require different retraction length due to stringing.

Patola, 2017-01-01:
I agree with you opinion. It won't help if you use firmware retraction though.

VBen, 2017-01-02:
perhaps a custom gcode per filament can be used to setup the firmware retraction

bubnikv, 2017-01-02:
Maybe a general way to override printer settings by filament settings would
be useful. At the prusa3d fork I have added a limit for a per filament
maximum volumetric extrusion rate, useful for multi material printing. I
can imagine a user interface similar to the object settings to add new
attributes to a filament.

lordofhyphens, 2017-01-25:
I've been thinking about override hierarchies myself (currently the idea for me is print -> filament -> printer for something that is shared), either for custom gcode or certain configurations that can logically be in different places.

Reusing the override interface makes some sense, but would require more refactoring to get them to save properly with the hierarchy.

alexrj, 2017-05-16:
It is now possible to override Retraction settings from Filament profiles :) Enjoy!

[image]

closed 2017-05-16 by alexrj

hoegge, 2018-01-24:
Any chance getting this implemented in Slic3r PE too?

lordofhyphens, 2018-01-24:
@hoegge ask in slic3r PE issue tracker.