slic3r/Slic3r #3644: Ooze prevention has ooze prevention settings combined instead of independent

Patola, 2017-01-01:
### Version
_Version of Slic3r used goes here_
1.2.9-607-gdf4f643
df4f6434dbdc1440f1dc14991606e4af21cff1b4

### Operating system type + version
Ubuntu 16.04 64 bits

### Behavior
In Print Settings → Multiple Extruders → Ooze prevention there are two settings:
"Enable", which adds a part-high skirt to be used for purging filament;
and "Temperature variation", which makes the inactive extruder(s) get to a different temperature so it does not keep oozing (this also prevents crystalization due to extended heating time).

* _Expected Results_

These settings should be independent. The user should be able to select the ooze shield and the temperature variation. Like that:
- Ooze Prevention -
Enable [X]
.. Temperature variation [ -60 ] Δ°C
.. Ooze shield [X]

* _Actual Results_
User cannot have temperature variation without the ooze shield although he can have the ooze shield without the temperature variation.

Note: this problem has an easy workaround, which is setting the standby temperature in the 'Tool change G-code' setting: M104 T[previous_extruder] S140 -- that's what I use, but it does not change the fact that there is no reason for the settings being combined.

lordofhyphens, 2017-01-25:
If you want to check to make sure that the two are okay in code (that it's just a UI issue), then you can comment out the lines around https://github.com/alexrj/Slic3r/blob/master/lib/Slic3r/GUI/Tab.pm#L876 in a dev build and see if it works for you.

I'd be very appreciative if you'd check that.

Patola, 2017-03-12:
@lordofhyphens it seems that I have skipped the message where you told me that, I am sorry. I think that at the time I was so busy I could not read all emails. I am looking into that, and it seems that just recently @alexrj added to this code with new logic (to which I had the observation that it might be undesirable, because single extruder people use the ooze shield feature too).

Jan-NiklasB, 2020-12-16:
Bump

This is an issue for me too, these settings must be independent in my opinion.
I don't want to waste a ton of material, where I only want my nozzel to be in standby while not in use.

Please fix it.