slic3r/Slic3r #3645: Commandline slic3r cannot handle spaces in filenames

thomasgresch, 2017-01-01:
### Version
1.31.6-Prusa

### Operating system type + version
OS X 10.11.5

### Behavior
The commandline version of slic3r cannot handle spaces in filenames:

`slic3r.pl Frangulin.stl --load /Users/tg/Library/Application Support/Slic3r/print/0.20mm NORMAL.ini -gui
`
returns:

`Cannot find specified configuration file (/Users/tg/Library/Application).`

the script assumes, that after the space another argument would come. I tried with single backslashes, tripple backslashes and quotes - all without success.

lordofhyphens, 2017-01-01:
You need to reproduce this in our Slic3r, not the Prusa fork (use their
issues page if you cannot reproduce with upstream Slic3r).

lordofhyphens, 2017-01-01:
Ah I also see what is wrong here. Put the path in quotes or escape the spaces with a leading \ for each one.

That is a shell thing, not Slic3r AFAIK.

lordofhyphens, 2017-01-01:
`--load "/Users/tg/Library/Application Support/Slic3r/print/0.20mm NORMAL.ini" `

lordofhyphens, 2017-01-01:
` --load /Users/tg/Library/Application\ Support/Slic3r/print/0.20mm\ NORMAL.ini`

closed 2017-01-01 by lordofhyphens

Mettlerfreund, 2018-06-15:
Using an 8.3 filename did the job for me. I had issues with this command on a windows machine:

slic3r-console -o .\some path\ --export-svg 3D_object.stl

lordofhyphens solution doesn't work here, but this works:

slic3r-console -o .\SOMEPA~1\ --export-svg 3D_object.stl 

See [https://en.wikipedia.org/wiki/8.3_filename](https://en.wikipedia.org/wiki/8.3_filename) 
Greetings M~