slic3r/Slic3r #3647: Cut dialog crashes on object with multiple shells

lordofhyphens, 2017-01-02:
### Version
1.3.0-dev 2be6e1a84f04e71a2033d2e141b58ca21e4f97ed

### Operating system type + version
Windows 10 x64, 32-bit perl (32bit ints)
Strawberry Perl 5.24

### Behavior
Selecting Cut on provided STL causes a hard crash of Slic3r. STL is manifold, but has 6 (I think intersecting) shells. Thanks Sketchup!
No config needed.

#### STL/Config (.ZIP) where problem occurs
[40kdalek1skp.zip](https://github.com/alexrj/Slic3r/files/680296/40kdalek1skp.zip)

lordofhyphens, 2017-01-02:
Found this by accident; splitting the object permits the cut dialog to start.

bubnikv, 2017-01-27:
@lordofhyphens  Great catch! I was begging our print guy Jindra for months to find an example to reproduce the crash on.

lordofhyphens, 2017-01-27:
Apparently the easiest way to get dodgy STLS is to have Sketchup in your workflow.