slic3r/Slic3r #3648: slic3r crashes when in GUI mode

g3gg0, 2017-01-02:
### Version
origin/master 16df4df8f87cb035528f4447a1a21ac79eb2e1af

### Operating system type + version
Windows 10 x64, 
perl 5, version 24, subversion 0 (v5.24.0) built for MSWin32-x86-multi-thread-64int

### Behavior
* load attached config
* add model from http://www.thingiverse.com/download:1156927 to buildplate
* slicer crashes with:
```
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
```

#### STL/Config (.ZIP) where problem occurs
config.ini (upload failed)

```
# generated by Slic3r 1.3.0-dev on 
avoid_crossing_perimeters = 0
bed_shape = 0x0,205x0,205x205,0x205
bed_temperature = 70
before_layer_gcode = M240
bottom_solid_layers = 4
bridge_acceleration = 0
bridge_fan_speed = 100
bridge_flow_ratio = 1
bridge_speed = 80
brim_connections_width = 0
brim_width = 0
complete_objects = 0
cooling = 1
default_acceleration = 0
disable_fan_first_layers = 2
dont_support_bridges = 0
duplicate_distance = 6
end_gcode = M220 S100\nG1 X0 F5000\nM104 S0 ; turn off temperature\nM140 S0 ; make sure the bed is turned off.\nG28 X0  ; home X axis\nM84     ; disable motors\n
external_fill_pattern = rectilinear
external_perimeter_extrusion_width = 0
external_perimeter_speed = 50%
external_perimeters_first = 0
extra_perimeters = 1
extruder_clearance_height = 20
extruder_clearance_radius = 20
extruder_offset = 0x0
extrusion_axis = E
extrusion_multiplier = 1.1
extrusion_width = 0.4
fan_always_on = 1
fan_below_layer_time = 60
filament_colour = #E9E9E9
filament_diameter = 1.7
filament_max_volumetric_speed = 0
filament_notes = ""
filament_settings_id = 
fill_angle = 45
fill_density = 100%
fill_pattern = rectilinear
first_layer_acceleration = 0
first_layer_bed_temperature = 110
first_layer_extrusion_width = 200%
first_layer_height = 0.105
first_layer_speed = 30
first_layer_temperature = 250
gap_fill_speed = 20
gcode_arcs = 0
gcode_comments = 0
gcode_flavor = reprap
has_heatbed = 1
infill_acceleration = 0
infill_every_layers = 1
infill_extruder = 1
infill_extrusion_width = 0
infill_first = 0
infill_only_where_needed = 1
infill_overlap = 15%
infill_speed = 60
interface_shells = 0
layer_gcode = 
layer_height = 0.105
max_fan_speed = 100
max_print_speed = 80
max_volumetric_speed = 0
min_fan_speed = 100
min_print_speed = 10
min_skirt_length = 3
notes = 
nozzle_diameter = 0.4
octoprint_apikey = 
octoprint_host = 
only_retract_when_crossing_perimeters = 1
ooze_prevention = 0
output_filename_format = [input_filename_base].gcode
overhangs = 1
perimeter_acceleration = 0
perimeter_extruder = 1
perimeter_extrusion_width = 0
perimeter_speed = 60
perimeters = 2
post_process = 
pressure_advance = 0
print_settings_id = 
printer_settings_id = 
raft_layers = 0
resolution = 0
retract_before_travel = 2
retract_layer_change = 0
retract_length = 1
retract_length_toolchange = 10
retract_lift = 0
retract_lift_above = 0
retract_lift_below = 0
retract_restart_extra = 0
retract_restart_extra_toolchange = 0
retract_speed = 40
seam_position = aligned
serial_port = COM5
serial_speed = 250000
skirt_distance = 6
skirt_height = 1
skirts = 3
slowdown_below_layer_time = 0
small_perimeter_speed = 15
solid_infill_below_area = 10
solid_infill_every_layers = 0
solid_infill_extruder = 1
solid_infill_extrusion_width = 0
solid_infill_speed = 20
spiral_vase = 0
standby_temperature_delta = -5
start_gcode = M117 Heating bed...\nM104 S180\nM190 S[first_layer_bed_temperature]\n\nM117 Heating extruder...\nM109 S[first_layer_temperature]\n\nG28\nG1 Z5 F5000\n\nM109 S[first_layer_temperature]\n\nM117 Cleaning...\nG1 X60 Y0\nG1 Z0 \nG4 P1000 \nG92 E0 \nG1 E30 \nG92 E0\nG4 P1000 \nG1 Z-0.5\nG1 E10\nG92 E0\nG1 Z0\nG1 X60 Y16.5 F2000\nG1 X60 Y20 F200\nG1 X50 F1000\nG1 X70 Y25\nG1 X60 Y30\n\nM117 Printing...
support_material = 1
support_material_angle = 0
support_material_contact_distance = 0.3
support_material_enforce_layers = 0
support_material_extruder = 1
support_material_extrusion_width = 0
support_material_interface_extruder = 1
support_material_interface_layers = 3
support_material_interface_spacing = 3
support_material_interface_speed = 100%
support_material_pattern = pillars
support_material_spacing = 3
support_material_speed = 50
support_material_threshold = 25
temperature = 240
thin_walls = 1
threads = 2
toolchange_gcode = 
top_infill_extrusion_width = 0
top_solid_infill_speed = 15
top_solid_layers = 4
travel_speed = 80
use_firmware_retraction = 0
use_relative_e_distances = 0
use_volumetric_e = 0
vibration_limit = 0
wipe = 0
xy_size_compensation = 0
z_offset = 0
```

lordofhyphens, 2017-01-02:
You need to put the ini file in a zip archive to upload it.

lordofhyphens, 2017-01-02:
Turn off background processing for starters, see if that avoids the crash. I can't reproduce on Linux :/

lordofhyphens, 2017-01-02:
I can't reproduce on my development Win32 system, same Perl version. I suspect it's one of the libraries on your system.

lordofhyphens, 2017-01-02:
[image]

g3gg0, 2017-01-02:
thanks for testing. how can i do e.g. a backtrace to find out whats causing the trouble?

about the .zip: already packed a zip with .stl and .ini but that failed to upload.

lordofhyphens, 2017-01-02:
Well nevermind, I just did manage to reproduce :D

g3gg0, 2017-01-02:
glad to hear :D

lordofhyphens, 2017-01-02:
It's in the infill generation, specifically "infill only where needed". Turn that off as a workaround.

g3gg0, 2017-01-02:
argh, misclicked. "close and comment" is not cancel...

lordofhyphens, 2017-01-02:
I'm not seeing the bad_alloc exception on my runs (could be because I'm running through powershell). Infill was recently rewritten by @bubnikv and then revised by @alexrj.

g3gg0, 2017-01-02:
can confirm that it depends how you start slic3r.pl
when starting from cmd.exe, it shows the error.

lordofhyphens, 2017-01-02:
I just ran it via command line, so it's not the GUI crashing. I wonder if it's a stack thing or if it's just a win32 thing (not 64bit). I edited your post to put an upload of the files.

bubnikv, 2017-01-02:
A "bad alloc" exception means you most likely ran out of the process
virtual space. A 32bit process on Windows can access only the lower 2GB of
a virtual address space, which is shared by all threads of a process and
every thread gets some stack space reserved. The virtual memory runs out
very quickly if too many threads get instantiated. Just use the process
explorer from sysinternals (now owned by Microsoft) to count the allocated
threads and to inspect their memory allocations, namely the reserved stack.
Often the application itself could allocate just around 1.2GB for itself
because of all the thread stacks reserved.

One can toggle one bit in the perl.exe header to tell the operating system
that the application is "large space aware", which then lets the process
use 3GB of virtual memory instead of just 2GB. This makes often a huge
difference. But one is better off to run a 64bit application on a 64bit
system with enough of physical RAM.

By the way, the 3D print path preview in Slic3r is a memory hog, it
requires around 3-4x more memory than just the slicing and G-code
generation. Instead of generating a list of triangles and quads, one shall
generate triangle and quad strips to reduce memory footprint.

lordofhyphens, 2017-01-02:
@bubnikv I was able to reproduce the problem on Linux x64, error message
there was a corrupted/double free.

`*** Error in `perl': double free or corruption (fasttop):
0x00007fc854a33e00 ***`

I kept an eye on the process memory usage using task manager and it only
reached around ~160MB before crashing.

It also doesn't crash if "Only infill where needed" is unchecked.

backtrace:
````
#0  0x00007fc884012067 in __GI_raise (sig=sig@entry=6) at
../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x00007fc884013448 in __GI_abort () at abort.c:89
#2  0x00007fc8840501b4 in __libc_message (do_abort=do_abort@entry=1,
    fmt=fmt@entry=0x7fc884145530 "*** Error in `%s': %s: 0x%s ***\n") at
../sysdeps/posix/libc_fatal.c:175
#3  0x00007fc88405598e in malloc_printerr (action=1,
    str=0x7fc884145718 "double free or corruption (fasttop)",
ptr=<optimized out>) at malloc.c:4996
#4  0x00007fc884056696 in _int_free (av=<optimized out>, p=<optimized out>,
have_lock=0) at malloc.c:3840
#5  0x00007fc882a883e6 in deallocate (this=<optimized out>, __p=<optimized
out>)
    at /usr/include/c++/4.9/ext/new_allocator.h:110
#6  deallocate (__a=..., __n=<optimized out>, __p=<optimized out>)
    at /usr/include/c++/4.9/ext/alloc_traits.h:185
#7  _M_deallocate (this=<optimized out>, __n=<optimized out>,
__p=<optimized out>)
    at /usr/include/c++/4.9/bits/stl_vector.h:178
#8  ~_Vector_base (this=0x7fc854a32e38, __in_chrg=<optimized out>)
    at /usr/include/c++/4.9/bits/stl_vector.h:160
#9  ~vector (this=0x7fc854a32e38, __in_chrg=<optimized out>) at
/usr/include/c++/4.9/bits/stl_vector.h:425
#10 ~IntersectionPoint (this=0x7fc854a32e08, __in_chrg=<optimized out>)
    at src/libslic3r/Fill/FillRectilinear.hpp:26
#11 ~pair (this=0x7fc854a32e00, __in_chrg=<optimized out>) at
/usr/include/c++/4.9/bits/stl_pair.h:96
#12 destroy (this=<optimized out>, __p=0x7fc854a32e00) at
/usr/include/c++/4.9/ext/new_allocator.h:133
#13 _M_destroy_node (this=0x7fc854a26e88, __p=0x7fc854a32de0) at
/usr/include/c++/4.9/bits/stl_tree.h:409
#14 std::_Rb_tree<long, std::pair<long const,
Slic3r::FillRectilinear::IntersectionPoint>, std::_Select1st<std::pair<long
const, Slic3r::FillRectilinear::IntersectionPoint> >, std::less<long>,
std::allocator<std::pair<long const,
Slic3r::FillRectilinear::IntersectionPoint> > >::_M_erase_aux (
    this=this@entry=0x7fc854a26e88, __position=..., __position@entry=...)
    at /usr/include/c++/4.9/bits/stl_tree.h:1867
#15 0x00007fc882a86f4a in erase (__position=..., this=0x7fc854a26e88)
    at /usr/include/c++/4.9/bits/stl_tree.h:874
#16 erase (__position=..., this=0x7fc854a26e88) at
/usr/include/c++/4.9/bits/stl_map.h:746
#17 Slic3r::FillRectilinear::_fill_single_direction
(this=this@entry=0x7fc854a240f0,
expolygon=...,
    direction=..., x_shift=x_shift@entry=0, out=out@entry=0x7fc85effcb00)
    at src/libslic3r/Fill/FillRectilinear.cpp:337
#18 0x00007fc882a87368 in Slic3r::FillRectilinear::_fill_surface_single
(this=this@entry=0x7fc854a240f0,
    thickness_layers=<optimized out>, direction=..., expolygon=...,
out=out@entry=0x7fc85effcb00)
    at src/libslic3r/Fill/FillRectilinear.cpp:398
#19 0x00007fc882a44826 in Slic3r::Fill::fill_surface (this=0x7fc854a240f0,
surface=...)
    at src/libslic3r/Fill/Fill.cpp:68
#20 0x00007fc882a7a953 in Slic3r::LayerRegion::make_fill
(this=0x7fc830196680)
    at src/libslic3r/LayerRegionFill.cpp:238
#21 0x00007fc8829d6484 in Slic3r::Layer::make_fills (this=0x7fc828001dc0)
at src/libslic3r/Layer.cpp:241
#22 0x00007fc882abf149 in operator() (a0=0x7fc828001dc0,
this=0x7fc85effced0)
    at /usr/include/boost/function/function_template.hpp:767
#23 Slic3r::_parallelize_do<Slic3r::Layer*>(std::queue<Slic3r::Layer*,
std::deque<Slic3r::Layer*, std::allocator<Slic3r::Layer*> > >*,
boost::mutex*, boost::function<void (Slic3r::Layer*)>)
(queue=0x7fc85e7fba50,
    queue_mutex=0x7fc85e7fb6e0, func=...) at src/libslic3r/libslic3r.h:85
#24 0x00007fc882abf069 in operator()<void (*)(std::queue<Slic3r::Layer*>*,
boost::mutex*, boost::function<void(Slic3r::Layer*)>), boost::_bi::list0> (
    f=@0x7fc8309ba788: 0x7fc882abf0d0
<Slic3r::_parallelize_do<Slic3r::Layer*>(std::queue<Slic3r::Layer*,
std::deque<Slic3r::Layer*, std::allocator<Slic3r::Layer*> > >*,
boost::mutex*, boost::function<void (Slic3r::Layer*)>)>, a=<synthetic
pointer>, this=0x7fc8309ba790) at /usr/include/boost/bind/bind.hpp:392
#25 operator() (this=0x7fc8309ba788) at
/usr/include/boost/bind/bind_template.hpp:20
#26 boost::detail::thread_data<boost::_bi::bind_t<void, void
(*)(std::queue<Slic3r::Layer*, std::deque<Slic3r::Layer*,
std::allocator<Slic3r::Layer*> > >*, boost::mutex*, boost::function<void
(Slic3r::Layer*)>),
boost::_bi::list3<boost::_bi::value<std::queue<Slic3r::Layer*,
std::deque<Slic3r::Layer*, std::allocator<Slic3r::Layer*> > >*>,
boost::_bi::value<boost::mutex*>, boost::_bi::value<boost::function<void
(Slic3r::Layer*)> > > > >::run() (this=0x7fc8309ba5d0) at
/usr/include/boost/thread/detail/thread.hpp:117
#27 0x00007fc88207daea in ?? () from
/usr/lib/x86_64-linux-gnu/libboost_thread.so.1.55.0
#28 0x00007fc8843900a4 in start_thread (arg=0x7fc85effd700) at
pthread_create.c:309
#29 0x00007fc8840c587d in clone () at
../sysdeps/unix/sysv/linux/x86_64/clone.S:111
````

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2017-01-02:
@bubnikv I was able to reproduce the problem on Linux x64, error message
there was a corrupted/double free.

`*** Error in 'perl': double free or corruption (fasttop):
0x00007fc854a33e00 ***`

I kept an eye on the process memory usage using task manager and it only
reached around ~160MB before crashing.

It also doesn't crash if "Only infill where needed" is unchecked.

backtrace:
````
#0  0x00007fc884012067 in __GI_raise (sig=sig@entry=6) at
../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x00007fc884013448 in __GI_abort () at abort.c:89
#2  0x00007fc8840501b4 in __libc_message (do_abort=do_abort@entry=1,
    fmt=fmt@entry=0x7fc884145530 "*** Error in `%s': %s: 0x%s ***\n") at
../sysdeps/posix/libc_fatal.c:175
#3  0x00007fc88405598e in malloc_printerr (action=1,
    str=0x7fc884145718 "double free or corruption (fasttop)",
ptr=<optimized out>) at malloc.c:4996
#4  0x00007fc884056696 in _int_free (av=<optimized out>, p=<optimized out>,
have_lock=0) at malloc.c:3840
#5  0x00007fc882a883e6 in deallocate (this=<optimized out>, __p=<optimized
out>)
    at /usr/include/c++/4.9/ext/new_allocator.h:110
#6  deallocate (__a=..., __n=<optimized out>, __p=<optimized out>)
    at /usr/include/c++/4.9/ext/alloc_traits.h:185
#7  _M_deallocate (this=<optimized out>, __n=<optimized out>,
__p=<optimized out>)
    at /usr/include/c++/4.9/bits/stl_vector.h:178
#8  ~_Vector_base (this=0x7fc854a32e38, __in_chrg=<optimized out>)
    at /usr/include/c++/4.9/bits/stl_vector.h:160
#9  ~vector (this=0x7fc854a32e38, __in_chrg=<optimized out>) at
/usr/include/c++/4.9/bits/stl_vector.h:425
#10 ~IntersectionPoint (this=0x7fc854a32e08, __in_chrg=<optimized out>)
    at src/libslic3r/Fill/FillRectilinear.hpp:26
#11 ~pair (this=0x7fc854a32e00, __in_chrg=<optimized out>) at
/usr/include/c++/4.9/bits/stl_pair.h:96
#12 destroy (this=<optimized out>, __p=0x7fc854a32e00) at
/usr/include/c++/4.9/ext/new_allocator.h:133
#13 _M_destroy_node (this=0x7fc854a26e88, __p=0x7fc854a32de0) at
/usr/include/c++/4.9/bits/stl_tree.h:409
#14 std::_Rb_tree<long, std::pair<long const,
Slic3r::FillRectilinear::IntersectionPoint>, std::_Select1st<std::pair<long
const, Slic3r::FillRectilinear::IntersectionPoint> >, std::less<long>,
std::allocator<std::pair<long const,
Slic3r::FillRectilinear::IntersectionPoint> > >::_M_erase_aux (
    this=this@entry=0x7fc854a26e88, __position=..., __position@entry=...)
    at /usr/include/c++/4.9/bits/stl_tree.h:1867
#15 0x00007fc882a86f4a in erase (__position=..., this=0x7fc854a26e88)
    at /usr/include/c++/4.9/bits/stl_tree.h:874
#16 erase (__position=..., this=0x7fc854a26e88) at
/usr/include/c++/4.9/bits/stl_map.h:746
#17 Slic3r::FillRectilinear::_fill_single_direction
(this=this@entry=0x7fc854a240f0,
expolygon=...,
    direction=..., x_shift=x_shift@entry=0, out=out@entry=0x7fc85effcb00)
    at src/libslic3r/Fill/FillRectilinear.cpp:337
#18 0x00007fc882a87368 in Slic3r::FillRectilinear::_fill_surface_single
(this=this@entry=0x7fc854a240f0,
    thickness_layers=<optimized out>, direction=..., expolygon=...,
out=out@entry=0x7fc85effcb00)
    at src/libslic3r/Fill/FillRectilinear.cpp:398
#19 0x00007fc882a44826 in Slic3r::Fill::fill_surface (this=0x7fc854a240f0,
surface=...)
    at src/libslic3r/Fill/Fill.cpp:68
#20 0x00007fc882a7a953 in Slic3r::LayerRegion::make_fill
(this=0x7fc830196680)
    at src/libslic3r/LayerRegionFill.cpp:238
#21 0x00007fc8829d6484 in Slic3r::Layer::make_fills (this=0x7fc828001dc0)
at src/libslic3r/Layer.cpp:241
#22 0x00007fc882abf149 in operator() (a0=0x7fc828001dc0,
this=0x7fc85effced0)
    at /usr/include/boost/function/function_template.hpp:767
#23 Slic3r::_parallelize_do<Slic3r::Layer*>(std::queue<Slic3r::Layer*,
std::deque<Slic3r::Layer*, std::allocator<Slic3r::Layer*> > >*,
boost::mutex*, boost::function<void (Slic3r::Layer*)>)
(queue=0x7fc85e7fba50,
    queue_mutex=0x7fc85e7fb6e0, func=...) at src/libslic3r/libslic3r.h:85
#24 0x00007fc882abf069 in operator()<void (*)(std::queue<Slic3r::Layer*>*,
boost::mutex*, boost::function<void(Slic3r::Layer*)>), boost::_bi::list0> (
    f=@0x7fc8309ba788: 0x7fc882abf0d0
<Slic3r::_parallelize_do<Slic3r::Layer*>(std::queue<Slic3r::Layer*,
std::deque<Slic3r::Layer*, std::allocator<Slic3r::Layer*> > >*,
boost::mutex*, boost::function<void (Slic3r::Layer*)>)>, a=<synthetic
pointer>, this=0x7fc8309ba790) at /usr/include/boost/bind/bind.hpp:392
#25 operator() (this=0x7fc8309ba788) at
/usr/include/boost/bind/bind_template.hpp:20
#26 boost::detail::thread_data<boost::_bi::bind_t<void, void
(*)(std::queue<Slic3r::Layer*, std::deque<Slic3r::Layer*,
std::allocator<Slic3r::Layer*> > >*, boost::mutex*, boost::function<void
(Slic3r::Layer*)>),
boost::_bi::list3<boost::_bi::value<std::queue<Slic3r::Layer*,
std::deque<Slic3r::Layer*, std::allocator<Slic3r::Layer*> > >*>,
boost::_bi::value<boost::mutex*>, boost::_bi::value<boost::function<void
(Slic3r::Layer*)> > > > >::run() (this=0x7fc8309ba5d0) at
/usr/include/boost/thread/detail/thread.hpp:117
#27 0x00007fc88207daea in ?? () from
/usr/lib/x86_64-linux-gnu/libboost_thread.so.1.55.0
#28 0x00007fc8843900a4 in start_thread (arg=0x7fc85effd700) at
pthread_create.c:309
#29 0x00007fc8840c587d in clone () at
../sysdeps/unix/sysv/linux/x86_64/clone.S:111
````

Tinchus2009, 2017-01-03:
I have been trying to see any pattern i the crashes. As I reported earlier, haveing the option checked sometimes crashes slic3r sometimes it doesn't. And I discovered this: the crash happens always when the infill is being calculated. BUT, it seems that the crash is not really produced by this option but something related to, MAY BE, calculations where no infill is made???? Because the real option that produces a crash, 100% of the times is when I set infill to 0%.
Setting infill to 0% produces a crash 100% of the times.

lordofhyphens, 2017-01-03:
@Tinchus2009 so does 100% infill.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

Salandora, 2017-01-03:
this problem also occurs with combine infill every x layers with x > 1.
And also depends on the extrusion width, had a model that slices well with everything set to 0.53 and crashes with 0.5, while nozzle diameter is 0.6.

I'm sorry I can't provide the stl file.
I hope this information helps in locating the bug.

g3gg0, 2017-01-03:
it crashe with no other model for me with these settings.

only the provided one crashed till today.

g3gg0, 2017-01-03:
did you do a run with valgrind yet?

lordofhyphens, 2017-01-03:
You are welcome to give it a shot ;), I haven't tried to run valgrind on a
static library  (libslic3r).

g3gg0, 2017-01-03:
<- windows only :(

Patola, 2017-01-10:
I see that there has been a lot of activity on this bug report, but I reported the same issue on Linux 11 days prior to this one. Should this one be marked as duplicate, or my own bug report, issue #3637 be marked as such?

lordofhyphens, 2017-01-10:
They are both linked, up to you at this point if you want to close yours or
not.

My personal preference is to use the one that either has more info in it or
is older.

Tinchus2009, 2017-01-16:
I just builted the latest dev version: 1.2.9-618-g1f920dc

I have tested this problem with different models, setting infill to 0% and to 100% is no longer causeing problems. But when activating "only infill where is nedded" I got some crashes, even with the same model. Teh same model with the same infill % sometimes crashed when activateing the option sometimes it didnt, no change at all between the test regarding the configuration. The only different thing I can look as "suspicious" is that the times it crashed, I was previously changeing parameters and slicein to test. After some succesfull slices I went back and activated the option and that times it crashed, so from my poor codeing habilities, could it the problem be that there is some variable not being cleared up that is causeing the problem?

lordofhyphens, 2017-01-16:
Could be related, did you try with a full restart of Slic3r between tests
to check that possibility  (that it might be something incorrectly set
during a restarted slice that wouldn't be in a clean system)?

Tinchus2009, 2017-01-16:
yes. tht is exactly wht im observeing. when it crashes, if i restart
slic3r, with no change un config at all, same object, the slice is
perfectly done

alexrj, 2017-03-06:
Fixed, thank you for the report.

closed 2017-03-06 by alexrj