slic3r/Slic3r #3649: First layer height used above support layers

dwillmore, 2017-01-02:
### Version
1.3.0-dev 1.2.9-507-g3a3b24e
### Operating system type + version
Fedora 24 current
### Behavior
In printing a knight from [http://www.thingiverse.com/thing:585218], I had my first layer height set to 0.3mm and my layer height to 0.1mm.  The head of the horse overhangs and needed support.  So, I enabled that.  To make sure it was good, I zoomed in on the area and noticed something odd.  It appears that the first layer of the model above the support was printed at the first layer height.  I would have expected it to be at the normal layer height as first layer height is only intended to improve bed adhesion.
[image]
[config.zip](https://github.com/alexrj/Slic3r/files/681273/config.zip)

I believe this is a bug.

#### STL/Config (.ZIP) where problem occurs
See the link to thingiverse above.

lordofhyphens, 2017-01-02:
That's bridging flow actually, so far as I can tell.
first layer 0.1
[image]

first layer 1.2
[image]

2x bridge flow
[image]

50% bridge flow
[image]

lordofhyphens, 2017-01-02:
Those layers are being printed with bridging flow, which is generally what you expect since there's a gap there, not the first layer.

If you can think of a reason we shouldn't be using bridging flow for this I'll reopen ;)

closed 2017-01-02 by lordofhyphens