slic3r/Slic3r #3653: Bridge width problem with nozzle diameter other than 0.5

Unsigno, 2017-01-03:
### Tested versions
1.2.9
1.3.0-dev

### Tested operating systems
Ubuntu 14.04 x64
Windows 7 x64

### Behavior
I am trying to slice a [x-end-idler for a PrusaI2](https://github.com/prusajr/PrusaMendel/blob/master/metric-prusa-lm8uu/x-end-idler.stl) , using 0.4 nozzle diameter. Which does not properly slice the inner part of bridges . 
The "Default extrusion width" is set to 0.5 and "Detect bridging perimeters" is enabled .
( After some tests, I am assuming that the "extrusion width" does not apply to the bridges, and instead uses the "nozzle diameter" )

### Slic3r v1.2.9 tests
**The result slicing with 0.5 nozzle diameter is perfect .**
[image]

**This is the result using 0.4 nozzle diameter.**
[image]

**I tried playing with some parameters, but just "nozzle diameter" has any effect . So i tested changing model measures to have rounded number of lines . But still not working .**
[image]

**[This calibration bridge](http://www.thingiverse.com/download:511664) with a 10mm width ( match with 24 lines , 1 less as expected ) its working**
[image]

###Slic3r v1.3.0-dev tests
**Reading about bridge issues i decided to test 1.3.0-dev version , that fix "Bridge 1" on original and scaled model , doing the first ( no perimeter ) line of the bridge attached to the perimeter and "forcing" the last line to overlap a bit . But "Bridge 2" still with holes .
This is the result with the original part , buts looks similar with scaled bridge .**
[image]

###Conclusion 
How I would calculate the width of a bridge, so that the fill is calculated correctly on v1.2.9 ??

Could you add the possibility of filling the bridge diagonally, to have a more precise number of lines ?? 

In some error cases the hole width is <1 line on the sides but >1 in total . Why no add a line centering whole bridge ??

bubnikv, 2017-01-03:
>How I would calculate the width of a bridge, so that the fill is calculated correctly on v1.2.9 ??

The issue we are facing is, there is no way to fill an arbitrary shape with an integer number of lines.

The extrusion placement is a NP-complete problem .
https://en.wikipedia.org/wiki/NP-completeness

The way Slic3r places bridges is following:
1) It finds spots, that need a support.
2) It inflates these spots.
3) It finds the best direction for stringing the bridges in a sense of supporting the ends of the bridges.
4) It fills the inflated spot with the direction calculated in the step 3).

In the step 4) the extrusions are placed, spaced to fill the spot calculated in 3), so the lines are not aligned with the bridged hole, but with the expanded hole. This is visible on your examples.

> Could you add the possibility of filling the bridge diagonally, to have a more precise number of lines ??

I don't understand how the diagonal placement would help. But what our print expert guy Jindra asks for is an override attribute to the automatic bridging direction. Namely, if the bridge could be supported in multiple directions, he wants to have a control over it. I think an override parameter with the modifier meshes would be useful.

>In some error cases the hole width is <1 line on the sides but >1 in total . Why no add a line centering whole bridge ??

One should probably change Slic3r to align the bridges to the unsupported spot, not to the inflated unsupported spot. But it could never be ideal. One may probably play with the extrusion width of the bridges (decrease, increase the flow slightly) to cover the area fully, but this needs experiments to evaluate the effects. Some slicers, for example the simplify3d, use a normal flow for the bridges, so the threads are thin depending on the layer thickness. Even it if toes not work ideally, it is a proof, that for bridges one may draw thinner threads than the nozzle diameter.

lordofhyphens, 2017-01-03:
Yeah, the extrusion placement problem is similar to placement and routing in VLSI and is NP complete as an ILP formulation.

lordofhyphens, 2017-01-03:
Well you can always draw thinner lines than the nozzle diameter, it's more that  the placement in xy is less reliable.  For bridging it works better because you are stretching the extrudate.

Unsigno, 2017-01-07:
First of all, thanks a lot for your quick answers and explanations, now i can understand how it works much better .

I'm not familiar with computational sciences and NP-complete terms, which was a good read . But I understand that the problem is an algorithm that serves for any bridge and not the filling of this (the simplest case) .

The diagonal placement sugestion was for fully fill the bridges if it can't do it rectilinear ( i see this on other slicer ), but it probably will drop lines and was for my misunderstanding . So we can ignore this comment .

I thought the "bridge flow ratio" was to adjust the flow at print time and had no effect on the slices. I have reduced the value to 0.7 and now it is completely filled .

> One should probably change Slic3r to align the bridges to the unsupported spot, not to the inflated unsupported spot. But it could never be ideal.

This is what I still do not understand. Why is the area inflated to support the bridge, which in many cases could be enlarged or reduced to fit, is prioritized to the bridge itself?

bubnikv, 2017-01-27:
> This is what I still do not understand. Why is the area inflated to support the bridge, which in many cases could be enlarged or reduced to fit, is prioritized to the bridge itself?

It was just simpler to implement that way.

The correct solution for the bridges would naturally be to extend the bridging region in the direction of the infill only. Something to put on my TODO list.