slic3r/Slic3r #3654: The program does not start on windows 10 x64

SergioRius, 2017-01-03:
### Version
1.2.9a

### Operating system type + version
Windows 10 64bit

### Behavior
* The program does not start. It stays showing empty controls and "does not respond" appears on the tittle. Either normal run or administrator run. Either 64 or 32-bit version.
* Tested on a clean, updated windows 10.
Download the windows zip, extract and double click on slic3r gui executable.

lordofhyphens, 2017-01-03:
Give the development build (see readme on main github page a try).

lordofhyphens, 2017-01-03:
https://bintray.com/lordofhyphens/Slic3r/slic3r_dev/1.3.0-dev

I have personally tested that on my Windows 10 x64 system.

Tinchus2009, 2017-01-05:
Im running latest development buil on wi10 64 with no problems. earlier version of dev versions also worked (apart from time to time of some built problem, but once builted it run perfectly on win10 64)

lordofhyphens, 2017-01-25:
No response from @SergioRius and I've linked the -dev builds. Closing.

SergioRius, 2017-01-25:
Why you closed the issue? You assumed that I'm a developer?
If you want me to try something, at last you should provide a link with downloadable binaries and instructions. I don't know what I was supposed to do with the above response.

FYI, I'm running a 4 screens environment with a vulkan (not opengl) brand new AMD card. I arrived to the conclusion that is something related to it.

lordofhyphens, 2017-01-25:
I did link the -dev build. 22 days ago. Extract the archive and run slic3r.bat. This is probably how Slic3r will be packaged from now on (or something very similar). 

There's a newer one on the page now.

lordofhyphens, 2017-01-25:
My first response was "Hey, check the -dev build, see the readme for more information". The build is linked from there and it's on the main page you'd have to browse to get here.

https://github.com/alexrj/Slic3r/blob/master/README.md

SergioRius, 2017-01-26:
Hi Joseph.
I tested the build downloadable at the link you gave and the program still doesn't open.
This is a capture of the program after launch (hope you can see it)
[image]

lordofhyphens, 2017-01-26:
99% sure it is something with your local system. Unfortunately, I would
basically need to be at your computer to debug this much further.

Do you have some kind of SLI thing going on?

OpenGL is a requirement for the plater.

alexrj, 2017-03-17:
@lordofhyphens Maybe this is caused by the manifest issue mentioned by bubnikv in another issue recently.

Tinchus2009, 2017-03-17:
Well, I havent had any issues related to this since some months. I have win 10 64b. Latest test: 5 min ago, I builted latest dev version, and Im already printing (by the way: great new options, 1.2.9 stable should definitly be replaced with any of latest dev versions!!)

lordofhyphens, 2017-03-20:
@alexrj Very unlikely; the manifest issue does not crop up when being run from perl.exe (which is what that build did). 

@SergioRius https://bintray.com/lordofhyphens/Slic3r/slic3r_dev/1.3.0-dev-6d0cb54#files Is there any difference with this build?

SergioRius, 2017-03-20:
I just downloaded it to desktop, unzipped and run. It does the same thing, the window opens and the stays blank. Tittle says not responding and the window can be resized or moved. When closing windows says the application is hung.
Closing a command line window that opens before the program closes it.

Does this program store any configuration in the system? It feels strange that after that time, upon opening the window goes to my 3th screen as I had it the first time i run it. Even when I erased the windows position cache with cleaners so many times. I feel could be related.

[image]

SergioRius, 2017-03-20:
Remember that I run an AMD gfx card with vulkan, not opengl.

lordofhyphens, 2017-03-20:
It probably is related to your GFX setup (meaning that unless you can build from source, figure out what is going on exactly and fix it, there isn't anything we can do). 

Slic3r keeps its profile directory in your user appdata folder on Windows, but there shouldn't be anything that prevents the program from opening in the first place there.

lordofhyphens, 2017-03-20:
@SergioRius Another test would be to check the prusa3d fork (their build system is slightly different from ours). I suspect it will still be broken.

vetefilete, 2017-03-25:
Hi!
I am having the same problem, but i have discovered that if you click on other tab when the progam is in the state of Sergio´s picture, the program load the plate and works fine if you don´t click on the 3D tab again. So I think there are a problem with 3D visualization. I am trying to find what differences are in my system between previous executations and now, because a few days before I have been printing and slicing without problem.

I hope this  data helps to fix the problem.

Regards!

lordofhyphens, 2017-03-28:
@SergioRius @vetefilete Please check again with new version from https://dl.slic3r.org/dev. 
A few things have changed very recently.

vetefilete, 2017-03-28:
@lordofhyphens I think this link is broken :/

lordofhyphens, 2017-03-28:
@vetefilete 
Oops, it's not https, but http. Fixed.
(https://bintray.com/lordofhyphens/Slic3r/slic3r_dev/1.3.0-dev-d5a0e78#files)

vetefilete, 2017-03-28:
@lordofhyphens I just tested it, I´ve downloaded the slic3r-master zip file and it does work fine.

Thank you very much Joseph!!

Zenzan007, 2017-06-07:
Hi Everyone, 
I'd like to add to this. Yesterday I was using slic3r without any hassles and then today, after installing Fusion 360 by Autodesk, Slic3r started misbehaving. Same situation as mentioned above with the program not responding etc. I grabbed the current Dev version, Slic3r-master.2017.06.04.1158.9f0335b.64bit, No change in behavior. 
By Selecting the 2D tab at the bottom and avoiding the 3D rendering it's behaving nicely. This was based on vetefilete's suggestion. 
If anyone has any other suggestion it would be greatly appreciated.

lordofhyphens, 2017-06-07:
If it is 3d that implies to me that fusion messed with your opengl
libraries.

SergioRius, 2017-06-07:
I also use another product from Autodesk. 3DStudio Max. But as far I know it uses the provided vulkan driver (program rendering config). In fact I clean reinstalled the gfx card drivers several times from the day I installed 3dsMax.
Games play Ok

Zenzan007, 2017-06-08:
So I've tried re-installing the driver, uninstalling the driver and rebooting the machine, installing old drivers and installing the latest greatest drivers. All to no avail. In the end I've settled for using Cura. 
It bugs me because it's admitting defeat. But My kossel mini is supported by Cura and the 3D rendering etc works.
Many thanks for you input Sergio and Lordofhyphens. 
I'll be sure to post if anything changes, I'll be hammering away at this in the mean time.

MANUELLAHUERTA, 2017-10-18:
If you still have problems, just try this new release, it works very well for me with W10!, in fact it solved all my troubles, much better than Cura, whatever the CAD:

https://github.com/prusa3d/Slic3r/releases

bubnikv, 2017-10-18:
The prusa3d fork uses a bit different way of rendering the 3D scene: It uses vertex buffer objects if available, and the variable layer height feature uses the OpenGL 2.0 shaders. As some customers had issues with it, I have implemented a "Help->System info" menu intem to print out all relevant information on the OpenGL driver. Independent of whether the prusa3d fork works for you or not, it may be useful to post the log generated by the prusa3d fork's System Info here to help with the debugging.

lordofhyphens, 2017-10-18:
@bubnikv I will probably steal that OGL information page. Looks handy for
collecting information about what could be going on with people's systems.

bubnikv, 2017-10-18:
@lordofhyphens You should steal the complete C++ rendering using the OpenGL vertex buffers. It makes a day and night difference on properly accelerated video cards.

animavitis, 2018-02-20:
Have same issue with not working slic3r on my Acer v3-771G laptop.
In my case slic3r try to use integrated graphics card, but you can still force an app to use the dedicated GPU. With NVIDIA, it’s really easy.

Open the NVIDIA control panel. Right-click on an empty area on your desktop and select the ‘NVIDIA Control Panel’ option from the context menu.

Go to the Desktop menu and enable the ‘Add “Run with graphics processor” to Context Menu’ option.

thats all:)

eltsov-da, 2018-10-09:
Have same issue. Prusa Edition work perfect.
HP ProBook 4540s Core i7, 8Gb RAM, SSD Intel HD 4000+AMD Radeon HD 7650M

robob3ar, 2019-04-10:
damn it, just had this same isse with new fresh install of win 10, updated and all, 3 gpus, latest nvidia drivers all .. what is wrong with this thing slic3r 1.41.3

lordofhyphens, 2019-04-10:
@robob3ar can't say a thing about the Prusa3d fork which is what you are using.

lordofhyphens, 2019-04-11:
I am unable to fix this anymore in Slic3r's codebase because I cannot reproduce this on my systems.

It is definitely a failure of OpenGL to initialize for Slic3r.

Apparently on Windows 10 and the NVlink tool you can assign which gpu gets used on startup.

SergioRius, 2019-04-11:
This does happen when the system has more than one screen, the main screen is not the first one and at some time in the past you moved totally or partially the slicer window to a secondary screen.
Although it opens in the main one it fails to render.
So, disconnect all the screens but the main one and reboot windows. Then open slicer and it will open an work in the main one. Then reconnect all other screens and don't move slicer window.

lordofhyphens, 2019-04-11:
Looking around, it doesn't look like there's a good fix for this on Windows.

robob3ar, 2019-04-12:
for some reason it only works properly on my main monitor - win 10 3 monitor setup, 1080ti as display card, 2x2080ti as render cards .. other option is to right click on the program and choose other open gl device..  this worked properly on win7

andrey-gvrd, 2020-09-03:
Same issue, Windows 10 using Ubuntu 18 via WSL. Intel(R) HD Graphics 620. No external monitors.

[image]