slic3r/Slic3r #3655: Choose "no" on "auto-start" print for Octoprint always start the print

gege2b, 2017-01-05:
hi all and happy new year :)

### Version
_Version of Slic3r used goes here_
Current dev ( 586d95b28a8e473bdcbd959f1e37d215ae531df5 )

### Operating system type + version

Kubuntu 16.60 x64

### Behavior
* _Describe the problem_
Sending Gcode to Octoprint, when prompted if the print should start as soon as the file is uploaded, I choose "no" but the print start anyway. It also start if I choose yes (as expected).

So basically, the print always start, no matter what I choose when prompted

* _Steps needed to reproduce the problem_

Load a model, slice it, send to printer, click 'no' for auto-start the print, look the print begin

Another question related to this new octoprint stuff : is there a way to disable this (prompting for the filename AND for autostart) ?

mloebl, 2017-01-08:
Same problem here too on Linux.  

I also wouldn't mind an option to disable that prompt to ask about auto starting.

gege2b, 2017-01-08:
OK at least, I'm not the only one who got this issue :)
For the option, I suggested something in #3657

mloebl, 2017-01-08:
@gege2b Nope :)  Meant to log this last week when I hit the issue, but forgot... remembered today, and saw you beat me to it.

gege2b, 2017-01-08:
aha :)

after digging a bit into the code, I think I got the culprit
Plater.pm line 1402 : 
 ```            print => $self->{send_gcode_file} ? 1 : 0,
should be : 
```            print => $self->{send_gcode_file_print} ? 1 : 0,

Still not tested, my octoprint server is not reachable right now

gege2b, 2017-01-09:
OK I can confirm (tested right now)
@alexrj or @lordofhyphens do you want me to do a PR for this tiny fix ?

lordofhyphens, 2017-01-09:
Go ahead.

closed 2017-03-18 by alexrj