slic3r/Slic3r #3656: Support generation may take long time, sometimes forever

alexxy, 2017-01-05:
### Version
all versions including latest git master
### Operating system type + version
Linux
### Behavior

Generation of support material especialy for two extruder setup may take very long time, sometimes forever. I attached example config and stl file (for that generating support may take up to few hours). If needed i can upload file for that support generation takes forever (i wait for 4 days and it didnt finished).

[trp-cage_w_sidechains.zip](https://github.com/alexrj/Slic3r/files/688194/trp-cage_w_sidechains.zip)

[pdb.zip](https://github.com/alexrj/Slic3r/files/688196/pdb.zip)

bubnikv, 2017-01-06:
I am currently fine tuning a new support generator. It is pretty fast, the support generation for this model took around 10 seconds with layer height 0.2mm. The code is available as the master of the prusa3d fork, but I do not consider it to be production ready yet, though the results are very promising.

[image]

The main issue with the original Slic3r support is the complexity of overhang curves, which get summed up, and the complexity of the clipper library boolean and mainly the offset functions, which have at best O(n log n) time complexity, but probably worse in some instances. Cura fights this problem by simplifying the contours by a non-shrinking algorihtm (filling in concave corners only), I am fighting the problem by a discretization into a regular grid, which is the way Simplify3D handles the problem. I will likely extend my code to stretch a shrink-wrap around the contours discretized into the regular grid to achieve smooth curves with low complexity.

bubnikv, 2017-01-06:
thanks for a great support torture test object anyway. If you have some more, please share them :-)

bubnikv, 2017-01-06:
By the way, for the following model the original Slic3r did not calculate supports in a whole day. One of the kiss of deaths is the "safety offset" done on a very complex curve. If the safety offset is performed curve after curve separately and then later merged, the complexity of the operation is greatly reduced. I did this change in the prusa3d fork (see the offset functions on ExPolygon and ExPolygons)

[Space_marine_365%.zip](https://github.com/alexrj/Slic3r/files/689498/Space_marine_365.zip)

alexxy, 2017-01-10:
I can upload object for which support generation takes forever (waited ~4 days and it didnt finished) if you interested in it =)

bubnikv, 2017-01-10:
Sure, please go ahead and provide the test case.

To give all the interested a peek into my progress, I am currently fighting with two major issues:

1) Too much supports are getting generated. Some kind of filtering is needed to remove supports from spots, which are overhangs, but the overhangs are too small to be worth the scars created by the supports. For example, with the bridging perimeters enabled one can print 70 degrees slopes with some level of success and if those overhangs are not too large, one is better off to leave them unsupported.

2) As usual, with every support implementation one needs to do the test prints to make sure the supports serve their purpose and they are easy to remove.

alexxy, 2017-01-10:
New example. I'll add one more that will take forever
[2l15.zip](https://github.com/alexrj/Slic3r/files/696629/2l15.zip)

alexxy, 2017-01-10:
Basicaly its a models of proteins and their structures.

alexxy, 2017-01-10:
This model takes forever
[Ribosome_v2.zip](https://github.com/alexrj/Slic3r/files/696651/Ribosome_v2.zip)

alexxy, 2017-01-10:
Last one is rendering of CryoEM electron density

sl42, 2017-01-10:
I have also a small test example from a friend where the slic3r support generation didn't finish.

[feder.zip](https://github.com/alexrj/Slic3r/files/697264/feder.zip)

Thank you very much for improving the support generation

alexxy, 2017-01-10:
I tryed git master version from prusa3d fork. But it seems doesnt generate support for me

bubnikv, 2017-01-11:
Thanks for the test objects.

I have created an issue at the prusa3d github fork. I will generate alpha builds likely today. The new support generator works and the supports are not that difficult to remove, but it generates too much of unnecessary supports to my taste.

Once I provide the prusa3d builds, please use the following issue for further communication about the testing of the new supports.
https://github.com/prusa3d/Slic3r/issues/91

bubnikv, 2017-01-11:
@alexxy what is 2l15 ? I am gonna print it, so I would like to know what I will own :-)

alexxy, 2017-01-11:
2l15 is a pdb code =) check it [here](http://www.rcsb.org/pdb/explore/explore.do?structureId=2l15)