slic3r/Slic3r #3657: Feature request : slight UI change for octoprint stuff

gege2b, 2017-01-06:
Hi all

Due to #3655 , I came with an idea to keep the new octoprint upload things (change the name, run the print after upload...) but still keep the old behaviour

I would suggest something like this screenshot, this will give the user the choice, and - IMHO - would not clutter too much the UI.

[image]

It's also possible to replace the "send to printer as..." button with an "Edit filename before upload" checkbox

Also, giving octoprint stuff its own place could be usefull in the future, if some other related features are added (i.e. basic control of the printer, or feedback of the current print, or anything else allowed by the octoprint API) - but that's not  the subject here

I know @alexrj is kind of reticent to add more options to the UI, but what do you think about that ? :)

cheers

gege2b, 2017-01-18:
Hi there
I'm just doing some UI test here : 

https://github.com/gege2b/Slic3r/tree/sendToPrinterUI

Right now it's not working (I still have to figure out how to take these checkboxes into account when clicking the send gcode button)
But it gave an idea of how it could look

I'm a real noob at perl programming, so please be indulgent :p

dwillmore, 2017-03-03:
I would welcome some setting somewhere that allowed the 'send to printer' to be customized to the extent it could optionally prompt for a new filename and/or start printing once uploaded.

Specifically, I would like the ability to *not* get a dialog for either of those features and just upload the file to the printer.

So, I see it as two check options.
1) Do/don't prompt for filename
2) Radio button:
 a) Do prompt for 'start printing after upload'
 b) Do not prompt for 'start printing after upload' and don't start printing anyway
 c) Do not prompt for 'start printing after upload' and start printing anyway.

I think that covers the normal cases.

gege2b, 2017-03-03:
Hi  @dwillmore 

that's exactly what the mod I made on my "sentToPrinterUI" branch does, but with only one checkbox ("print [or don't] after upload") and two buttons ("send" and "send as..."). 

Doing this, there isn't any popup (except if you click on "send as" OR if a file with the same name already exists on Octoprint), it also covers the normal cases

I think there is less overhead in the UI this way (one checkbox and two buttons instead of one button and 4 check/radio), but its something pretty subjective I admitt

still waiting to know the dev's opinion on this feature request/piece of crappy code :)

dwillmore, 2017-03-03:
Hey, @gege2b.  Thanks for your work on this!  I often have files of the same name on the server and I want it to overwrite them without an extra prompt--but I can see where people might want to be prompted about that.  

I'm not as concerned with the complexity of the setting screen as I am with not breaking peoples workflows and/or making their workflow simpler.  Maybe that's just me being a Unix person--I don't mind lots of options to thing as long as I can tune it to work the way I want and end up with a simiple tool.

Does that make sense from a UX perspective?  One off things can be complex, but everyday tasks should be as simple as possible.

gege2b, 2017-03-03:
I don't mind having a lot of options too, but I know @alexrj prefer to minimize them as much as possible. And I too wish to get rid of these popups, hence this PR

Here's what I've actually done (my last message was referring to an old try witch is obsolete now) : 
[image]

maybe renaming the second checkbox "override existing file without asking" (or something like that) would be more explicit, and will smooth the workflow by avoiding every popups

it seems we are the only two person interested by this feature BTW  ^_^

dwillmore, 2017-03-03:
Ahh, I see the difference in what we are thining.  I was thinking in the normal UI there would just be the 'send to printer' button next to the normal 'export stl' and 'export gcode' buttons.  The options I was thinking of would be in Printer Settings->OctoPrint Upload.

Does that make more sense?

gege2b, 2017-03-03:
OK I see your point. and indeed It make sense for a kind of "permanent" option. 

When I first tinker about this options, I choose to have them on a per upload basis, so they can be changed on the fly more easily. That's why they're in the Plater tab

They can be saved too (i.e. the same way than the options in the preferences dialog) to make them remembered between two slicing sessions. But I didn't looked too much into this code yet

lordofhyphens, 2017-03-04:
Yeah, sorry about the relative lack of velocity. I don't use octoprint myself.

dwillmore, 2017-03-04:
@lordofhyphens don't sweat it.  It's just us chickens here. ;)  Maybe @gege2b and I can get this to the point of a pull request.

gege2b, 2017-03-04:
yep!  
@lordofhyphens  the reason of this try is to get the "old style" behaviour, before the auto-print and rename-prompt feature. 

Don't misunderstand me, they are great features, but I think there should be a way for users to enable/disable them if needed.
That's what I propose here, with @dwillmore suggestions, trying to clutter the UI as less as I can

The real problem here is that I'm new at perl programming, and I'm pretty sure there are better ways to do things, but it "just work" right now

gege2b, 2017-03-15:
@dwillmore , I just did a PR #3765 based on your suggestions (moving options to printer tab)

Default options mimic the actual behaviour, feel free to test :)

alexrj, 2017-03-18:
Closing - see my comment in the PR. We now have a spool dialog. One click, but better UX. Sorry.

closed 2017-03-18 by alexrj