slic3r/Slic3r #3658: Advanced->Extrusion width is computed over layer height

msrd0, 2017-01-08:
### Version

local/slic3r-bin 1.2.9-3

### Operating system type + version

ArchLinux Rolling Release

### Behavior

When I open Advanced tap on Print Settings and hover my mouse over any text field in the Extrusion Width fieldset, the tooltip tells me that a percentage will be computed over layer height. This doesn't make much sense to me (tell me if I'm wrong). The layer height can be adjusted for all layers and for the first on on the Layers and Perimeters tap. It would make more sense to me if the default extrusion width would be computed over nozzle diameter and the others over the default width.

alexrj, 2017-01-09:
Width-over-height (WoH) has been a meaningful parameter for a while. Changing the semantics of percent extrusion width would instantly break tons of existing configurations.

Patola, 2017-01-09:
As a curiosity...
I am writing a 3d printing book in brazilian portuguese and I am teaching tons of stuff from slic3r, cura and simplify3d. I have made a explicit recommendation for readers to avoid using percentages because it is indeed very counterintuitive to use the layer height. I strongly agree with @msrd0 that it is much more sensible for the percentages to be relative to the nozzle diameter. But I also know that it would instantly break all sorts of things, and I do not think it is worth the effort to change that now.

Salandora, 2017-01-09:
Maybe a little checkbox in the preferences could help here.
If checked it is relative to layer height. 
If unchecked then relative to nozzle diameter.
With default to checked?
That would not break anything, but gives the option to choose it yourself.

I also would like to have it as relative to nozzle diameter, so I can switch between Printer Settings without constantly adapting the extrusion width.

lordofhyphens, 2017-01-25:
This isn't something we can change at this point. 

If you are using explicit extrusion widths, then you should know what you are doing. Otherwise, let Slic3r pick in general and override if necessary to get the toolpaths to meet your engineering needs.

closed 2017-01-25 by lordofhyphens

7eggert, 2018-08-16:
Percentage of layer height is useless for me as I know I set e.g. 0.1 mm. Printing the first layer at 0.2 mm using a 0.5 mm nozzle (which I will change on occasion) is not in my book of sane things - but I tried it anyway and it worked ...

If you add %L or %N (auto-changing % to %L on display and back on saving, %Nozzle_width can be integrated without breaking existing configurations. (I came to github today to propose this feature.)

lordofhyphens, 2018-08-16:
@7eggert please don't propose features in closed issues.